AndyQ/NFCPassportReader #1: Package is out of date

myurieff, 2019-06-21:
Hello. 

I'm getting this error when trying to build the project:

`
Showing Recent Messages
: the Package.resolved file is most likely severely out-of-date and is preventing correct resolution; delete the resolved file and try again
`

Any ideas? Thanks.

AndyQ, 2019-06-21:
Check what version of Xcode you are running. You need to be on Beta 2. 

I've had reports of this happening when on Beta 1.

dshepherd, 2019-06-21:
I get the same error on Beta 2 (build 11M337n)

AndyQ, 2019-06-21:
Interesting - I  checked out the project to a clean folder and built with no issues.  Suspect its just a beta bug.  Closing for now as not sure what I can do about this but will keep an eye out!

closed 2019-06-21 by AndyQ

eldewy, 2019-06-30:
I also run the same issue. 
I run it on XCode 11.0 Beta 2 and MacOS Catalina 10.15 Beta
I manually went in my xcodeproj - file and deleted the Package.resolved file. When I rebuild it it creates the file newly but the same error occurs. 
I'm super curious to try out your project and would be happy if we could resolve this issue together.

AndyQ, 2019-06-30:
You could try removing the included NFCPassportReader override item (From the top of the project).

Then, select File menu - Swift Packages - Resolve Package Versions.
That will then resolve the remote package to the latest current version.
Once that works (do a clean and rebuild) then you re good to go.

If you want, you can then re-drag in the NFCPassportReader folder back into the project so you can edit the package code.