AndyQ/NFCPassportReader #5: No information

advatar, 2019-07-05:
Just started playing with this and get this log in the console

```
tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2826fce20>)
2019-07-05 13:07:02.143849+0200 NFCPassportReaderApp[3327:1102765] [CoreNFC] 00000002 808f0500 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): C06AEF680289C5B215A5A44F8B436C518641680C
Take the most significant 16 bytes to form the Kseed
	Kseed: C06AEF680289C5B215A5A44F8B436C51
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: C06AEF680289C5B215A5A44F8B436C5100000001
	Calculate the SHA-1 hash of D
		Hsha1(D): 7E70C4819710ECBFE199B53ECA24E30F6A2844AF
	Form keys Ka and Kb
		Ka: 7E70C4819710ECBF
		Kb: E199B53ECA24E30F
	Adjust parity bits
		Ka: 7F70C4809710ECBF
		Kb: E098B53ECB25E30E
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: C06AEF680289C5B215A5A44F8B436C5100000002
	Calculate the SHA-1 hash of D
		Hsha1(D): 00E55AD96F1EC5E179B1676E77DC796F19B5A5A8
	Form keys Ka and Kb
		Ka: 00E55AD96F1EC5E1
		Kb: 79B1676E77DC796F
	Adjust parity bits
		Ka: 01E55BD96E1FC4E0
		Kb: 79B0676E76DC796E
DATA - [163, 226, 37, 23, 62, 202, 242, 13]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: A3E225173ECAF20D
Generate an 8 byte random and a 16 byte random
	RND.IFD: 4E1B5C8FABC350C9
	RND.Kifd: 5842FCEDF71298DBC91817A6FDD4289E
Concatenate RND.IFD, RND.ICC and Kifd
	S: 4E1B5C8FABC350C9A3E225173ECAF20D5842FCEDF71298DBC91817A6FDD4289E
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: 52551BB50E3A810CD41D897D3DC701FBBA7AFD9962C310E145252431EB63744B
x0: 52551BB50E3A810C
y0: A7D44DF6F4158315
x1: D41D897D3DC701FB
y1: 170284638F1041C0
x2: BA7AFD9962C310E1
y2: 129D2A638CEF5C9A
x3: 45252431EB63744B
y3: 2BD51054FAAB6E46
x4: 8000000000000000
y4: 90933F700050255A
y: 90933F700050255A
bkey: 79B0676E76DC796E
akey: 01E55BD96E1FC4E0
b: 497A8D0860359C36
a: 1F2CE3C9895BBD52
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: 1F2CE3C9895BBD52
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: 52551BB50E3A810CD41D897D3DC701FBBA7AFD9962C310E145252431EB63744B1F2CE3C9895BBD52
Error reading tag: sw1 - 63, sw2 - 00 - reason: No information given
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 0.)
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}
```

Any pointers to what could have gone wrong? It is a Swedish passport btw.

AndyQ, 2019-07-05:
I'd say that your MRZ Key is in correct.

If the number is shorter than 9 chars and includes < characters, then you need to include the < characters in the key (the checksum should come out the same).

advatar, 2019-07-05:
passportNumber 89332616 dateOfBirth 040531 expiryDate 200210
mrzKey 89332616404053172002101

That is what I have. Looks ok to me. I noticed however that secureMessaging has not been set here: https://github.com/AndyQ/NFCPassportReader/blob/master/Sources/NFCPassportReader/TagReader.swift#L208

advatar, 2019-07-05:
so, it should really be 89332616< then I guess. Testing

advatar, 2019-07-05:
That < did the trick but now I get disconnected:

```
019-07-05 19:27:12.278648+0200 NFCPassportReaderApp[3938:1385928] [CoreNFC] 00000002 81cf5d80 -[NFCTagReaderSession transceive:tagUpdate:error:]:706  Error Domain=NFCError Code=100 "Tag connection lost" UserInfo={NSLocalizedDescription=Tag connection lost}
Error reading tag - Tag connection lost
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 0.)
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}

```

advatar, 2019-07-05:
Cool, got it to work now! Thge trick seems to be to just put the phone on the passport and not move it.

advatar, 2019-07-05:
Working on a MRZ scanner now. Will submit PR when it is done.

AndyQ, 2019-07-06:
Glad that sorted the problem. Will add this to the docs!

closed 2019-07-06 by AndyQ