AndyQ/NFCPassportReader #6: Not working in device

falsharabati, 2019-07-09:
I'm just trying to run the code, it's working in the simulator but in a real device, it's crash with below log.
I face this issue also [](https://github.com/AndyQ/NFCPassportReader/issues/1#issuecomment-507065292)

`dyld: Symbol not found: _$s7SwiftUI19UIViewRepresentableP13_layoutTraits3forAA07_LayoutF0VSg0C4TypeQz_tFTq
  Referenced from: /var/containers/Bundle/Application/EF99674A-73B5-4DB7-800E-3FDFBA3B32A1/NFCPassportReaderApp.app/NFCPassportReaderApp
  Expected in: /System/Library/Frameworks/SwiftUI.framework/SwiftUI
 in /var/containers/Bundle/Application/EF99674A-73B5-4DB7-800E-3FDFBA3B32A1/NFCPassportReaderApp.app/NFCPassportReaderApp
(lldb) `

AndyQ, 2019-07-09:
What version of iOS are you running - probably only works with latest Beta (Beta 3)

falsharabati, 2019-07-09:
Yes, I just now upgraded to Beta 3 and its working with me.
thank you

falsharabati, 2019-07-09:
But i get error
`2019-07-09 20:20:34.887679+0300 NFCPassportReaderApp[314:6517] [AXRuntimeCommon] Unknown client: NFCPassportReaderApp
`

Any advice?

AndyQ, 2019-07-09:
Doing what and at what point?

falsharabati, 2019-07-09:
Direct after launch the app

AndyQ, 2019-07-09:
Sorry unless you have more information (other errors, etc) I have no idea.

falsharabati, 2019-07-10:
Now its working fine thank you :)

closed 2019-07-10 by falsharabati

anguyen1, 2019-12-19:
@falsharabati, 
Hello, I am running into similar issue [AXRuntimeCommon] Unknown client.
Can you please share what you did to address this issue?
Thank you,
Anh

AndyQ, 2020-05-03:
Could you paste the full output log when you run the app.