AndyQ/NFCPassportReader #9: Error while building

Shweta25111990, 2019-07-29:
i am getting below issue:
Showing Recent Messages : the Package.resolved file is most likely severely out-of-date and is preventing correct resolution; delete the resolved file and try

i am running the project on xcode 11 beta 4.Tried downloading XCode 11 beta 2 , but could not succeed.

Could you help me?

atullis-scandit, 2019-07-29:
This happens when you add a dependency twice. Remove NFCPassportReader from the project navigator and it should work

AndyQ, 2019-07-31:
Try deleting the included swift package from the project (NOT the one at the bottom of the explorer view but the one at the top), then rescanning the swift dependancies.  That may fix the issue.

AndyQ, 2019-08-28:
Closing as no more info

closed 2019-08-28 by AndyQ