AndyQ/NFCPassportReader #10: Error session.connect( to: NFCTag).

SaliienkoAleksandr, 2019-07-30:
I always have an error when I am trying to connect to tag. I couldn't find any information how to resolve this problem.
I tried on Xcode 11 beta 4 and iOS 13 beta 4, Xcode 11 beta 5 and iOS 13 beta 5
Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x2800d7000 {Error Domain=nfcd Code=28 "Tag Not Found" UserInfo={NSLocalizedDescription=Tag Not Found}}}

AndyQ, 2019-07-31:
Can you provide more information?  

Do you have the ePassport application ID set in your info.plist?
Are you just tapping the passport or are you holding your phone against the passport until it finishes reading (it takes between 8-15 seconds to read the passport).  If you move it away too soon then it loses the connection.

SaliienkoAleksandr, 2019-07-31:
I'm trying to scan my Ukrainian ePassport.
I add UID A0000002471001, NFCEntitlements with two keys:
NFC Data Exchange Format,
NFC tag-specific data protocol.
Then create NFCTagReaderSession(pollingOption: .iso14443, delegate: self) -> begin it.
Delegate method find tag, but when I'm trying to connect session to tag I always receive such error:

tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x28120a130>)
2019-07-31 15:05:11.359110+0300 NFCPassportReaderApp[828:47880] [CoreNFC] 00000002 83214d00 -[NFCTagReaderSession _connectTag:error:]:553  Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x281285590 {Error Domain=nfcd Code=28 "Tag Not Found" UserInfo={NSLocalizedDescription=Tag Not Found}}}
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}

Here is my delegate method code:

public func tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag]) {
        
        Log.debug( "tagReaderSession:didDetect - \(tags[0])" )
        if tags.count > 1 {
            session.alertMessage = "More than 1 tags was found. Please present only 1 tag."
            return
        }
        
        let tag = tags.first!
        var passportTag: NFCISO7816Tag
        switch tags.first! {
        case let .iso7816(tag):
            passportTag = tag
        default:
            session.invalidate(errorMessage: "Tag not valid.")
            return
        }
     
        // Connect to tag
        session.connect(to: tag) { [unowned self] (error: Error?) in
            if error != nil {
               
                session.invalidate(errorMessage: "Connection error. Please try again.")
                return
            }
            
            self.readerSession?.alertMessage = "Authenticating with passport....."

            self.tagReader = TagReader(tag:passportTag)

            self.startReading()

        }
    }

SaliienkoAleksandr, 2019-07-31:
> Can you provide more information?
>
> Do you have the ePassport application ID set in your info.plist?
> Are you just tapping the passport or are you holding your phone against the passport until it [...]

AndyQ, 2019-07-31:
That sounds like you are successfully finding and connecting to the NFC Tag in the passport, but for some reason it loses connection to the tag.

Try holding your phone to different areas of the passport to find the strongest signal- e.g Front page, back page, open passport and hold to different pages.

I sometimes get that when I put my phone on the wrong area of a passport or there are too pages between the phone and the chip.

SaliienkoAleksandr, 2019-07-31:
I tried all day with different Ukrainian passports, but always had the same error.=((
You don't have such problem?

AndyQ, 2019-07-31:
Nope - one other thing to check - make sure your MRZ key is correct!

If the passport number in the MRZ Zone has an < in it then you need to include that'd character in the MRZ key!

E.g. Bottom line of MRZ:
LC123456<4GBR8106268M2206308<<<<<<<<<<<<<<<

Then the MRZKey would be LC123456<481062682206308

The checksum works out the same if the < is missed off (as that value is converted to 0 on checksum calculation!) BUT the key includes it!

SaliienkoAleksandr, 2019-07-31:
Yes, I know, I tried with this '<' and without and nothing helped.
But we first begin session, then in delegate method we find tag and try connect session to tag, and after, we have to try read tag where we use this MRZ key for authentication. 
Is it right?!
I have error when I just found tag and trying to connect session to tag, before I'm trying to read tag.

AndyQ, 2019-08-01:
Can you print out the error in the session.connection(to:tag) ?
e.g.
        session.connect(to: tag) { [unowned self] (error: Error?) in
            if error != nil {
                print( "Error connecting to tag - \(error!)" )
                session.invalidate(errorMessage: "Connection error. Please try again.")
                return
            }
......

May show a little more info.

SaliienkoAleksandr, 2019-08-01:
Error connecting to tag - Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x282295560 {Error Domain=nfcd Code=28 "Tag Not Found" UserInfo={NSLocalizedDescription=Tag Not Found}}}

AndyQ, 2019-08-01:
It sounds like it maybe either an Apple bug in the connect API or your passport isn't correctly responding to the connect request for some reason (signal strength?) as the chip is detected OK but it just can't connect.

Unfortunately there is nothing I can do if this is the case as this is all within the Apple APIs.  May be worth raising a bug request with them!

Just one last thing to check - you don't have any other devices with NFC tags in close proximity do you?  E.g. your passport isn't sitting on a MacBook or on an iPad when you try to read it?  As this can confuse the reader!.  You can check this by putting your passport flat on an empty part of a table and placing your phone on top when starting the read.

SaliienkoAleksandr, 2019-08-01:
I think it could be the bug of Apple, because we had Error Domain, I found that NFCErrorDomain: The domain for errors associated with Core NFC APIs. I tried to scan me passport on android application and everything is OK.
Interesting that I used your application to scan passport on beta 2 (or beta 1) in June and I could connect to tag. 
Now when I try to do this on beta 4, beta 5 I have such errors on all versions of application (June version, where it was ok, and last version on git).
I have already wrote to the Apple Feedback Assistant about this bug.

AndyQ, 2019-08-01:
That is most odd - I haven't seen this issue and so far all the passports I've been reading have read OK with all the betas. 

I'm closing this issue as I don't think its anything that I can fix, but if you get any more information then please add it here, or if you do think its an issue I need to fix please let me know!

closed 2019-08-01 by AndyQ

SaliienkoAleksandr, 2019-08-01:
Ok, thanks.