AndyQ/NFCPassportReader #16: Tag connection Lost

fbenedet, 2019-10-01:
The passport is italian, when i try to scan i receive this error:

tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2815545a0>)
2019-10-01 13:42:36.797989+0200 NFCPassportReaderApp[3356:752688] [CoreNFC] 00000002 835cd200 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): 94DABEBC97DA3236463B7FF71ADD120253FCAD34
Take the most significant 16 bytes to form the Kseed
	Kseed: 94DABEBC97DA3236463B7FF71ADD1202
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: 94DABEBC97DA3236463B7FF71ADD120200000001
	Calculate the SHA-1 hash of D
		Hsha1(D): 8D15C29688D0D5EDAF2DB94307B47A5706C874C9
	Form keys Ka and Kb
		Ka: 8D15C29688D0D5ED
		Kb: AF2DB94307B47A57
	Adjust parity bits
		Ka: 8C15C29789D0D5EC
		Kb: AE2CB94307B57A57
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: 94DABEBC97DA3236463B7FF71ADD120200000002
	Calculate the SHA-1 hash of D
		Hsha1(D): 7F6F3B84ACF542303C1E93784FEE5082EA0B8BE1
	Form keys Ka and Kb
		Ka: 7F6F3B84ACF54230
		Kb: 3C1E93784FEE5082
	Adjust parity bits
		Ka: 7F6E3B85ADF44331
		Kb: 3D1F92794FEF5183
2019-10-01 13:42:37.207573+0200 NFCPassportReaderApp[3356:752688] [CoreNFC] 00000002 835cd200 -[NFCTagReaderSession transceive:tagUpdate:error:]:726  Error Domain=NFCError Code=100 "Tag connection lost" UserInfo={NSLocalizedDescription=Tag connection lost}
Error reading tag - Tag connection lost

ebrienza, 2019-10-01:
I have no problem with my Italian passport

fbenedet, 2019-10-01:
Do you have changed something or you use the same code in the master
repository?

ebrienza, 2019-10-01:
Just GUI modifications, but I've done just some test some version ago.

fbenedet, 2019-10-01:
OK thanks

fbenedet, 2019-10-01:
anyway it is strange that with my passport the application give me an error when call the method: 
        tag.sendCommand(apdu: toSend) {...}
Error:
Optional(Error Domain=NFCError Code=100 "Tag connection lost" UserInfo={NSLocalizedDescription=Tag connection lost})

pcehb, 2019-10-07:
I am having the exact same error but with my British passport. Same method. About 60% of the time it gives: Error reading tag: sw1 - 63, sw2 - CF - reason: Unknown error - sw1: 99, sw2: 207, and the other 40% it gives the Tag connection lost error.

dileepa88, 2019-10-16:
I'm testing with a Malaysian passport and getting the same error in 
'doMutualAuthentication()'

AndyQ, 2019-10-16:
I'm not sure why some people are getting the Tag connection lost quite often. I've had it sometimes but its usually because the signal between the passport and the phone is weak - usually either not quite aligned right, or sometimes too much distance/obstructions between the phone and passport chip.

Oddly though, I've had a streak where this would happen a lot and then all of a sudden corrects itsself and everything just works.  Unfortunately, the actual NFC communication bit is totally a black box so I have no real way of seeing why - and Apple as yet provide no low level debugging info.

Not sure though what If anything I can do about this sadly!

I'll leave this open for a little bit and if anyone has more info please post!

AndyQ, 2019-11-12:
Closing now as no more info

closed 2019-11-12 by AndyQ

wellbranding, 2022-07-07:
@fbenedet  Hello, we as well facing issues with the Italian passport. Can you check if yours is this one?
[https://www.consilium.europa.eu/prado/en/ITA-AO-02003/index.htm](https://www.consilium.europa.eu/prado/en/ITA-AO-02003/index.html)
Have you resolved the issue?