AndyQ/NFCPassportReader #18: Guide to running the example

rooi, 2019-10-14:
Hi,

Could you please provide a guide for building and running the example? I tried using iOS 13, but the example app does not seem to respond the inputs. Did I build it wrong, or am I using the incorrect inputs?

Thanks!

Roy

mickeyl, 2019-12-04:
Same problem here. After entering the details, nothing happens and I can't seem to get the 'Scan Passport' button(?) to appear. Moreover, what's the purpose of the UISegmentedControl?

leo1988joseph, 2019-12-14:
Hi @mickeyl , any progress yet? I am in desperate need of the code now. Could you guide me?

mickeyl, 2019-12-15:
I didn't pursue this further, I hope @AndyQ will chime in. That said, did you check whether the Info.plist contains the necessary tag IDs for passport reading? After doing what https://stackoverflow.com/a/56516232/415982 told, reading (not decrypting though) worked in my example program.

AndyQ, 2019-12-15:
Something changed in one of the iOS 13 beta which stopped textfield updating the model.
This was fixed in 1.0.5 (just realised I hadn't pushed the podspec! - done!).

Hopefully now the UI should correctly update (you need at least 8 chars for the passport number and valid dates for both - yymmdd format).

If you are still having issues please let me know!

leo1988joseph, 2019-12-16:
Hi, we have been given an expired passport to test with form the client. I
just want to know wether the code could actually fetch the data from the
NFC chip of passport. Which line in the code should I comment so that I can
at-least know the values of the passport are read correctly? (iOS Swift)

Thanks and regards
Leo Joseph

AndyQ, 2019-12-16:
@leo1988joseph All passports are treated equally - from a NFC Reading perspective, there is no difference between an expired passport and an non-expired passport.  It is down to the implementor to check if a passport has expired and handle that accordingly.

It may be that the passive authentication may fail if the certificate used to sign the passport had also expired (or is no longer included in the master list) but thats not necessarily the case (but again the implementor needs to explicitly check for this)

AndyQ, 2020-01-08:
Closing as assuming this is now fixed

closed 2020-01-08 by AndyQ

leo1988joseph, 2020-02-12:
Hi AndyQ,

While using the NFCReader in some of the US passports, we are getting a
flickering screen as shown in the video attached. Can I please know were
exactly in the code you have written to not read certain kind of passports?
It din't read some Canadian passports too. Kindly help.

Thanks and regards
Leo Joseph

AndyQ, 2020-02-12:
Hi Leo,  there is nothing in the code that blocks certain passports.  There may be (are) incompatibilities with some passports - especially some older ones,as it appears that there were many ways of implementing the e-passport chips.

Unfortunately there is no video attached so I can't see what your issue was.  If you could raise a new issue for this that would be great.

leo1988joseph, 2020-02-12:
Hi Andy,

This is the video I was saying.

Please have a look. Just see is this issue worth a new bug?

Regards
Leo Joseph

On Wed, Feb 12, 2020 at 7:02 PM Leo Joseph <leojoseph1988@gmail.com> wrote:

> Hi AndyQ,
>
> While using the NFCReader in some of the US passports, we are getting a
> flickering screen as shown in the video attached. Can I please know were
> exactly in the code you have written to [...]

AndyQ, 2020-02-12:
Hi Leo, unfortunately as this issue is closed I don't think attached files get added

leo1988joseph, 2020-02-12:
How can i forward the video to you?

Can you give me some other way by which i can forward it to you?

AndyQ, 2020-02-13:
Can you please open a new issue?

leo1988joseph, 2020-02-13:
I am getting some error while creating a new issue.

Can you for now give me some medium by which I can just share the video to
show what I meant by flickering? Its bit urgent.

Regards

AndyQ, 2020-02-13:
You can send me a the video directly - my email is in my profile