AndyQ/NFCPassportReader #19: Chinese passport showed NFCError Code=200

univer2012, 2019-10-16:
hello author,
I want to identify my Chinese passport with CoreNFC in iOS 13 by using your this project. But when I input passport Num,Date of birth and Passport expiry date and tap the scan button, it output the error on the console:

```
Build DO'8E
	DO8E: 8E08057D0E40499415A3
Construct and send protected APDU
	ProtectedAPDU: 0CA4020C1587090136D7EC32E9FADAD78E08057D0E40499415A300
[SM] <NFCISO7816APDU: 0x280715e00>
Error reading tag: sw1 - 6A, sw2 - 82 - reason: File not found
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}
```
And sometimes the error is that:
```
NFCPassportReaderApp[9722:648432] [CoreNFC] 00000002 82754700 -[NFCTagReaderSession transceive:tagUpdate:error:]:710  Error Domain=NFCError Code=100 "Tag connection lost" UserInfo={NSLocalizedDescription=Tag connection lost}
Error reading tag - Tag connection lost
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}
```

I really want to know what happen,and next what should i do. Please tell me the steps that solve these problems.

AndyQ, 2019-10-16:
Unfortunately, there isn't enough there to go on!

At what point to you get the error - do you get any of the tags read?

The Tag connection lost error is usually because the connection between the passport and the phone gets dropped. This happens for many reasons - none of which I can actually see why and as yet Apple don't provide any low level debugging info.

AndyQ, 2019-10-29:
Closed as no info provided

closed 2019-10-29 by AndyQ