AndyQ/NFCPassportReader #20: could not install pods and run example

zealit, 2019-11-01:
When I try to $ pod install in the Example directory, I get the message:

[!] CocoaPods could not find compatible versions for pod "OpenSSL-Universal/Framework":
  In Podfile:
    NFCPassportReader (from `https://github.com/AndyQ/NFCPassportReader.git`) was resolved to 0.0.10, which depends on
      OpenSSL-Universal/Framework

None of your spec sources contain a spec satisfying the dependency: `OpenSSL-Universal/Framework`.

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

Note: as of CocoaPods 1.0, `pod repo update` does not happen on `pod install` by default.

When I start the workspace in Xcode, it shows errors with missing files in Pods/Target Support Files/

AndyQ, 2019-11-01:
What version of Cocoapods are you using?  And have you done a pod repo update?

zealit, 2019-11-05:
yes solved now.

- but the example is missing the file PassiveAuthentication.swift. 
- also problems in Passport.swift: model.passportDataElements => passportDataElements is inaccessible due to private protection level
- PassiveAuthentication(masterListURL) => use of unresolved identifier PassiveAuthentication

AndyQ, 2019-11-05:
The latest version has undergone a number of changes.  PassiveAuthentication has moved into he NFCPassportModel now and can be automatically done when reading the passport (by setting the masterListURL on the passportReader object (see sample app).

Also, the NFCPassportModel now exposes the data fields directly so you shouldn't need to access the passportDataElements field any more!

Note - my sample app no longer uses Passport.swift and this will be removed from the next version (it should have been gone already but I missed that!)

closed 2019-11-05 by AndyQ