AndyQ/NFCPassportReader #21: Passport showed NFCError Code=200 (DEUTSCHLAND)

canssamet, 2019-11-08:
Hello there
Thank you very much for preparing such a beautiful package. And of course, congratulations. I'd be happy to say we used it seriously.
I want to talk about a problem written by someone else and get help. You can view the closed topic from the link below. Actually, I'd like to continue with that. But it's closed. Therefore, the answer you specify there is not the correct answer.

We want to solve the problem that we have identified but cannot solve with your help. We can reach the solution with the speed that you will gain when you develop the package.
In fact, the causes of this problem in reality are;

The same passport was failing on android devices. **The reason is that the dg12 file is not in that passport**. If you do not have that file on the Android side of the other file, but I was able to switch to the ios side. [DOCATCH] gets error even though I do not. I don't know where to fix it, please help.

AndyQ, 2019-11-08:
How far through the read do you get?  Does it fail immediately or do you get the COM, SOD or any datagroups read?

Also, are you using the latest version - there have been a large number of changes in this release which should hopefully help.

AndyQ, 2019-11-12:
I've fixed a small bug where the read was failing under certain circumstances. Please let me know of 1.0.1 fixes your problem.

canssamet, 2019-11-12:
> How far through the read do you get? Does it fail immediately or do you get the COM, SOD or any datagroups read?
> 
> Also, are you using the latest version - there have been a large number of changes in this release which should hopefully help.

Thanks for your answer. I've updated the latest version. Now work. There is only one error in the dg3 file (Error reading tag: sw1 - 69, sw2 - 82 - cause: Security status not provided) but it works. I do not use a DG3 file. If the dg3 file is working, everything will be fine. : DD. Thanks again

AndyQ, 2019-11-12:
You won't be able to read the DG3 item as you don't have the security certificates but the code should skip over that now.  If it doesn't please could you add your log to this:
https://github.com/AndyQ/NFCPassportReader/issues/22

closed 2019-11-12 by AndyQ