AndyQ/NFCPassportReader #23: tagReaderSessionDidBecomeActive() not being called

atatlisu, 2019-11-13:
Hi,

   I use pods for the library. The nfc pop-up appears on the screen however when i hold a Turkish passport or Turkish ID Card (which has similar nfc chip) , the methods tagReaderSessionDidBecomeActive() or tagReaderSession() not being called.

I configured info.plist, entitlements and capabilities also checked it from apple developer account.
However ReadID just works fine.

I also tried to use sample app but, it also didn't work.

Am i missing anything ? Can you give me some clue about the issue ?

Thanks and regards...

atatlisu, 2019-11-14:
Update: I successfully read my passport using sample app, however still in my app  not works. tagReaderSessionDidBecomeActive() or tagReaderSession() not being called.

atatlisu, 2019-11-14:
Update2: When i call ContentView via 
```
let passportDetails = PassportDetails()
            
            self.navigationController?.pushViewController(UIHostingController(rootView: ContentView().environmentObject(passportDetails)),
                                                          animated: false)
```
it works but, when i call PassportReader directly like this

```
let reader = PassportReader()
            let masterListURL = Bundle.main.url(forResource: "masterList", withExtension: ".pem")!
            reader.setMasterListURL( masterListURL )
            let mrzKey = "S02189603395081962008181"
            reader.readPassport(mrzKey: mrzKey, completed: { (model,error)  in
                if(error != nil){
                    print("error in passport: \(error?.localizedDescription)")
                }else{
                    print("model: \(model?.nationality)")
                }
            })
```

it doens't work. The methods tagReaderSessionDidBecomeActive() or tagReaderSession() inside the PassportReader not being called.

AndyQ, 2019-11-14:
Have you added the necessary items to your info.plist - specifically the necessary privacy items, the ISO7816 application identifiers for NFC Tag Reader Session, and also the required entitlements?

atatlisu, 2019-11-15:
i set them both, please look at update2 they are in the same app, same viewcontroller but first code block working however the second one is not.

AndyQ, 2019-11-15:
Just a thought - try making PassportReader a class property - if you are creating it in the method you are calling it from, then you may find its being deallocated before it finishes.

e.g.
`class ScanNFCPassportViewController: UIViewController {
    let passportReader = PassportReader()

    func readPassport( mrzKey: String ) {
        let masterListURL = Bundle.main.url(forResource: "masterList", withExtension: "pem")!
        passportReader.setMasterListURL(masterListURL)
        passportReader.readPassport(mrzKey: mrzKey, completed: { [unowned self] (passportModel, error) in
            ...
    }
}
`

atatlisu, 2019-11-15:
When i init the the PassportReader as a global variable the issue is fixed, thank you ...

closed 2019-11-15 by atatlisu