AndyQ/NFCPassportReader #24: Error performing active authentication

JanMisker, 2019-11-14:
When running the example app, I get my data from the passport. However the authentication fails.
```
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 63, sw2 - 00 - reason: No information given
BAC Failed
Starting Basic Access Control (BAC)
BAC Successful
Reading tag - COM
DG Found - ["DG1", "DG2", "DG15", "DG3", "DG14"]
Reading tag - SOD
Reading tag - DG1
Reading tag - DG2
Reading tag - DG15
Reading tag - DG14
Performing Active Authentication
Error reading tag: sw1 - 67, sw2 - 00 - reason: Wrong length
IN CALLBACK
error 20 at 0 depth lookup:unable to get local issuer certificate
```
Quick debug shows that something goes wrong in the `send` in `doInternalAuthentication`, maybe the `expectedResponseLength` should be different?

This is with a passport from The Netherlands.

BTW Great project! Since a few days it is possible in the digital ID app of The Netherlands to verify yourself by reading your passport in their app, so I wanted to know how it's done :)

AndyQ, 2019-11-14:
Ah yes - you may want to try 1.0.4 - I've just found out today that older passports won't return more than 160 bytes per read, and prior versions were trying to read the max 255 bytes.

1.0.4 adds a fix so that if we get the Wrong length error, then I re-establish BAC and drop the data read amount to 160 bytes (that works on my older passports).

If you still get this error then please reduce the amount in TagReader::reduceDataReadingAmount and let me know what ended up working for you.

JanMisker, 2019-11-15:
I still get the same error, and I also placed a breakpoint in `TagReader::reduceDataReadingAmount` but it's never called.

The error seems to originate from the send in `TagReader::doInternalAuthentication`.

AndyQ, 2019-11-15:
OK, could you replace the TagReader::doInternalAuthentication with the following:

```
    func doInternalAuthentication( challenge: [UInt8], completed: @escaping (ResponseAPDU?, TagError?)->() ) {
        let randNonce = Data(challenge)
        
        let data : [UInt8] = [0x00, 0x88, 0x00, 0x00, 0x08] + challenge + [0x00]
        let cmd = NFCISO7816APDU(data:Data(data))!

        //let cmd : NFCISO7816APDU = NFCISO7816APDU(instructionClass: 00, instructionCode: 0x88, p1Parameter: 0, p2Parameter: 0, data: randNonce, expectedResponseLength: 50)
        send( cmd: cmd, completed: completed )
    }

```

That worked with my test passport.  Basically rather than reading in 50 bytes, I'm not not setting a length to read (you need to do this manually rather than using the main constructor as that won't let you pass in a zero length).

Let me know if that works for you.

JanMisker, 2019-11-15:
Still the same... I set logLevel to .debug and uncommented some logging, I get this:
```
Performing Active Authentication
Mask class byte and pad command header
	CmdHeader: 0C88000080000000
Pad data
	Data: 267B4A1651E7FB3C8000000000000000
Encrypt data with KSenc
	EncryptedData: 31AD8A5933A330209DDAF67EDE760667
...
Construct and send protected APDU
	ProtectedAPDU: 0C8800001D87110131AD8A5933A330209DDAF67EDE7606678E082580D51D542E510F00
[SM] <NFCISO7816APDU: 0x281676eb0>
[SM] [153, 2, 103, 0, 142, 8, 51, 193, 78, 220, 135, 60, 171, 70], sw1:0x67 sw2:0x00
Error reading tag: sw1 - 67, sw2 - 00 - reason: Wrong length
```

AndyQ, 2019-11-15:
Little stumped but going through the JMRTD code, you could try:

        let data : [UInt8] = [0x00, 0x88, 0x00, 0x00, 0x00, 0x00, 0x08] + challenge + [0x00,0x00]

Guessing a little here though - bit tricky to test remotely.

For info, this is based on case 4e - https://github.com/ugochirico/SCUBA/blob/master/scuba_smartcards/src/main/java/net/sf/scuba/smartcards/CommandAPDU.java)

Any better?

smlu, 2019-12-01:
The `internalAuthentication` command for eMRTD (as far as I tested) requires DO97 to be present in the command APDU. The `Le` field in DO97 should be set to either variable signature length (`0x00`) or to an actual signature length. 
When I tested setting the `Le` to an actualy signature length (RSA - 0x80) it worked with no problem.  The problem that I encountered was when trying to use `0x00` for `Le`. The class `NFCISO7816APDU` doesn't like `0` for `Le` field instead `256` should be used. But I couldn't use `256` with this library because it wrongly encodes `256` as `0x100` instead of `0x00` in DO97.
Maybe this is also the problem here.

@JanMisker could you test this PR #28 and see if it fixes your problem?

AndyQ, 2019-12-10:
Could you please test against 1.0.5 - this has the above PR merged in.

puneetmahali, 2019-12-12:
I am also facing the same error-
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 63, sw2 - 00 - reason: No information given
BAC Failed

Also, I used the latest version 1.0.5 and try to make it lot of changes also but not successful. I am trying with German Passport.
Do you have any idea about it?

smlu, 2019-12-13:
@puneetmahali the `sw=6300` error usually means that external authentication (establishment of BAC session) failed. Based on your log I'm assuming this is the case here. This usually happens when wrong parameters were used to establish BAC session with passport eg.: wrong passport no., wrong format for date of birth and date of expiry (should be `YYMMDD`) etc... Could you double check you've entered correct parameters in correct field and the input data matches the data in MRZ?

puneetmahali, 2019-12-13:
Thanks for your quick help @smlu . Yes, you are right, i just enter the full year like 1992 but the format is like enter the last two digits only which means 92.

- So, now I am able to read the German passports and not another countries. How can i read the another countries as well.

- From where I can achieve the X.509 certificate like in this project @AndyQ using masterList.pem certificate with openssl. Is there need to buy the access or we can use free of that.
Because I am getting error log - "error 20 at 0 depth lookup:unable to get local issuer certificate"
Also in UI Not Genuine red cross logo because certificate is expired.

so, Please help me more about the X.509 certificate and read about the another countries passport. Also I want to read the ID's as well which approach I need to use for that.

Thanks in advance.

AndyQ, 2019-12-13:
The error 20 at 0 depth lookup is probably because you are trying to use the included masterList.pem file (which is actually just a single self generated test per certification included to stop build errors).

If you look in the scripts folder, you will see instructions for getting a valid masterList.pem file.

Regarding reading other countries passports - these should work - if you are getting the same sw=6300 error, then its likely that the mrz key parameters are not right (see the troubleshooting section in the readme for possible solutions).

puneetmahali, 2019-12-13:
Thank you so much @AndyQ for quick reply.
Where is the Scripts folder. Can you please guide me lil bit more to check it out.
Yes, the Brazilian Passport takes one extra character for Passport number like if the actual passport number is xxxxxxx then need to enter xxxxxxxx**<** and this is < the extra chracter. Why we need to enter this extra character?

Also, Is it necessary to include the .pem file with X.509 certificate?

AndyQ, 2019-12-13:
The scripts folder is in the root of the repo. 
There is a reader file in there that explains the process. (Note I cant provide a pre-generated masterList.pem file). 

I have no idea why you need to include the additional < characters - that just the way it is a documented in the spec.

AndyQ, 2019-12-15:
Closing this now as looks like the original issue is fixed

closed 2019-12-15 by AndyQ