AndyQ/NFCPassportReader #25: SM data objects incorrect / Wrong length

maxxx777, 2019-11-16:
```
Starting Basic Access Control (BAC)
BAC Successful
Reading tag - COM
got resp - ResponseAPDU(data: [95, 1, 4, 48, 49, 48, 55, 95, 54, 6, 48, 52, 48, 48, 48, 48, 92, 6, 97, 117, 103, 107, 108, 109], sw1: 144, sw2: 0)
DG Found - ["DG1", "DG2", "DG7", "DG11", "DG12", "DG13"]
Reading tag - SOD
Error reading tag: sw1 - 6C, sw2 - 0E - reason: Wrong length Le: SW2 indicates the exact length - (exact length :14)
ERROR - Wrong length Le: SW2 indicates the exact length - (exact length :14)
Reading tag - SOD
Error reading tag: sw1 - 69, sw2 - 88 - reason: SM data objects incorrect
ERROR - SM data objects incorrect
```
then
```
Starting Basic Access Control (BAC)
BAC Successful
Reading tag - SOD
Error reading tag: sw1 - 6C, sw2 - 0E - reason: Wrong length Le: SW2 indicates the exact length - (exact length :14)
ERROR - Wrong length Le: SW2 indicates the exact length - (exact length :14)
Reading tag - SOD
Error reading tag: sw1 - 69, sw2 - 88 - reason: SM data objects incorrect
ERROR - SM data objects incorrect
```
repeats endlessly till the session expired

Affected version: `1.0.4`
Document: `ID card from Kazakhstan`

**Fix the issue**

adding `self.tagReader?.reduceDataReadingAmount()` inside the condition 
```
} else if errMsg == "SM data objects incorrect" {
  // Can't read this element security objects now invalid - and return out so we re-do BAC
  completed(nil)
}
```
fixes the issue:
```
} else if errMsg == "SM data objects incorrect" {
   // Can't read this element security objects now invalid - and return out so we re-do BAC
   self.tagReader?.reduceDataReadingAmount()
   completed(nil)
}
```

AndyQ, 2019-11-16:
Thanks will be fixed in next version

maxxx777, 2019-11-17:
> Thanks will be fixed in next version.
> 
> Also, looking, I think a slightly better fix (with less opportunity for unexpected side effects) would be to change the bit below to:
> ` } else if errMsg.hasPrefix( "Wrong length" ) { // Should now handle errors 0x6C xx, and 0x67 0x00`

Nope, it fails. 
Even if `} else if errMsg.hasPrefix( "Wrong length" )` moved before `} else if errMsg == "SM data objects incorrect" {`.

AndyQ, 2019-11-17:
Hmm thats a little weird as the order shouldn't really matter.

If I'm reading your initial flow right - you first get the "Wrong length Le..." (which wasn't handled), so we retry the read, and get the SM data objects incorrect.  
You then reduce the data amount to be read and redo BAC which succeeds, and then you retry reading the SOD which then works.

What should happen wit the fix is we handle the "Wrong length Le" command, and this should reduce the data amount to be read, redo the BAC and then retry the SOD (just missing out the additional read which gets the SM error.

Could you check and verify that is what's happening or paste the debug log?

Thanks very much

maxxx777, 2019-11-24:
`BAC failed` with changing 
`} else if errMsg == "Wrong length" {`
to 
`} else if errMsg.hasPrefix( "Wrong length" ) {`

```
Starting Basic Access Control (BAC)
BAC Successful
Reading tag - COM
got resp - ResponseAPDU(data: [95, 1, 4, 48, 49, 48, 55, 95, 54, 6, 48, 52, 48, 48, 48, 48, 92, 6, 97, 117, 103, 107, 108, 109], sw1: 144, sw2: 0)
DG Found - ["DG1", "DG2", "DG7", "DG11", "DG12", "DG13"]
Reading tag - SOD
Error reading tag: sw1 - 6C, sw2 - 0E - reason: Wrong length Le: SW2 indicates the exact length - (exact length :14)
ERROR - Wrong length Le: SW2 indicates the exact length - (exact length :14)
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 6A, sw2 - 81 - reason: Function not supported
BAC Failed
```

AndyQ, 2019-11-25:
Could you turn on debug logging (In Logging.swift, change
    logLevel : LogLevel = .info
to
    logLevel : LogLevel = .debug

And paste the log?

maxxx777, 2019-11-26:
```
tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x280587210>)
Starting Basic Access Control (BAC)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): <value>
Take the most significant 16 bytes to form the Kseed
	Kseed: <value>
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: <value>
	Calculate the SHA-1 hash of D
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
		Kb: <value>
	Adjust parity bits
		Ka: <value>
		Kb: <value>
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: <value>
	Calculate the SHA-1 hash of D
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
		Kb: <value>
	Adjust parity bits
		Ka: <value>
		Kb: <value>
DATA - [138, 72, 16, 252, 36, 108, 134, 199]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: <value>
Generate an 8 byte random and a 16 byte random
	RND.IFD: <value>
	RND.Kifd: <value>
Concatenate RND.IFD, RND.ICC and Kifd
	S: <value>
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: <value>
x0: <value>
y0: <value>
x1: <value>
y1: <value>
x2: <value>
y2: <value>
x3: <value>
y3: <value>
x4: <value>
y4: <value>
y: <value>
bkey: <value>
akey: <value>
b: <value>
a: <value>
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: <value>
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: <value>
DATA - [6, 252, 68, 104, 159, 22, 250, 23, 78, 81, 68, 6, 75, 96, 15, 17, 10, 70, 144, 132, 36, 150, 143, 20, 139, 204, 49, 73, 194, 52, 255, 223, 39, 98, 186, 83, 29, 85, 134, 234]
Decrypt and verify received data and compare received RND.IFD with generated RND.IFD <value>
Calculate XOR of Kifd and Kicc
	Kseed: <value>
	Concatenate Kseed and c
		D: <value>
	Calculate the SHA-1 hash of D
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
[... 359 lines cut ...]
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
		Kb: <value>
	Adjust parity bits
		Ka: <value>
		Kb: <value>
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: <value>
	Calculate the SHA-1 hash of D
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
		Kb: <value>
	Adjust parity bits
		Ka: <value>
		Kb: <value>
Error reading tag: sw1 - 6A, sw2 - 81 - reason: Function not supported
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 0.)
BAC Failed
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}
```

`<value>` replaces real values just to hide them in public. please let me know if you need some of real values.

AndyQ, 2019-11-27:
Hmmm, Im still a bit confused as to why you fix worked but my adjustment didn't (as as far as I can see its actually doing the same thing unless there is an additional read that I'm missing!).

I'll do some testing with your fix and make sure that it doesn't break anything that I can easily see and put that in place.

Thanks for all your help with this!

AndyQ, 2019-12-10:
Could you please test with 1.0.5?

maxxx777, 2019-12-15:
It fails in `1.0.5`.

https://github.com/AndyQ/NFCPassportReader/issues/25#issuecomment-557930459

smlu, 2019-12-16:
@maxxx777 could you try setting var [maxDataLengthToRead](https://github.com/AndyQ/NFCPassportReader/blob/ccb7a7940df514b2c8e4973aa59ed8b8024ac739/Sources/NFCPassportReader/TagReader.swift#L180) to `224` or `256` and see if that changes anything?

maxxx777, 2020-01-03:
@smlu yep, all values for `maxDataLengthToRead` up to `0xE7` work. Does it mean this type of ID has smaller size of data blocks?

smlu, 2020-01-07:
As far as I can tell some passports can't handle larger block sizes that's why you have to request smaller data chunk.
 Btw the `224` was taken from JMRTD library: [https://sourceforge.net/p/jmrtd/code/HEAD/tree/trunk/jmrtd/src/main/java/org/jmrtd/PassportService.java#l225](https://sourceforge.net/p/jmrtd/code/HEAD/tree/trunk/jmrtd/src/main/java/org/jmrtd/PassportService.java#l225). 
Could you also try set var to `256` (`0x100`) and see if it works?

maxxx777, 2020-01-10:
I guess it won't work with `256` as we need to pass `UInt8` value to command https://github.com/AndyQ/NFCPassportReader/blob/b6ef7cab05ddfee06b8ef197c4aad5d906cabf89/Sources/NFCPassportReader/TagReader.swift#L278

smlu, 2020-01-15:
@maxxx777 You are right. I have slightly modified version of this library so the above suggestion works.
Ok, so what you can try for test is setting the constant `cmd` in [TagReader.swift:L280](https://github.com/AndyQ/NFCPassportReader/blob/b6ef7cab05ddfee06b8ef197c4aad5d906cabf89/Sources/NFCPassportReader/TagReader.swift#L280) to
```
       NFCISO7816APDU(
            instructionClass: 00,
            instructionCode: 0xB0,
            p1Parameter: offset[0],
            p2Parameter: offset[1],
            data: Data(),
            expectedResponseLength: 256
        )
```
What argument `expectedResponseLength: 256` does is telling passport that arbitrary long chunk of data is expected in response.

AndyQ, 2020-01-16:
I'm currently testing the above fix (successfully so far but have some more passports I want to run through it), and if its consistent good I'll update the main repo.

maxxx777, 2020-01-17:
> @maxxx777 You are right. I have slightly modified version of this library so the above suggestion works.
> Ok, so what you can try for test is setting the constant `cmd` in [...]

this change works as well 👍

AndyQ, 2020-02-05:
I've been testing the above and all is working on my passports - it will in the next release!

closed 2020-02-05 by AndyQ