AndyQ/NFCPassportReader #26: NFC Scan timeouts without scan

arunlj561, 2019-11-22:
I have generated the MRZKey. when start to Scan Details it opens the NFC view, the method gets invoke - "tagReaderSessionDidBecomeActive".
After a while it timeouts. 
"invalidate - Session timeout ". 
master.pem file is also generated as mention in the scripts folder
May I know what I am missing.

AndyQ, 2019-11-26:
Are you using the sample app or your own and if its your own, have you got the correct Plist settings set to allow the NFC chip to find the correct application ids?

Which passport are you using and are you sure its a biometric enabled passport?

Try holding the upper part of the phone to different parts of the passport - ideally, find out roughly where the chip is embedded, open the passport to that page.

Do you get any info in the debug logs (turn on debug logging in Logger.swift).

arunlj561, 2019-11-26:
You can close the issue it's working, was missing the tag reader to be mention in info.plist.

closed 2019-11-26 by arunlj561