AndyQ/NFCPassportReader #27: [German ID] Conditions of use not satisfied

maxxx777, 2019-11-26:
Affected version: `1.0.4`
Document: `ID card from Germany`

Info Logs:
```
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 69, sw2 - 85 - reason: Conditions of use not satisfied
BAC Failed
```

Debug Logs:
```
tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x28393a070>)
Starting Basic Access Control (BAC)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): <value>
Take the most significant 16 bytes to form the Kseed
	Kseed: <value>
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: <value>
	Calculate the SHA-1 hash of D
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
		Kb: <value>
	Adjust parity bits
		Ka: <value>
		Kb: <value>
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: <value>
	Calculate the SHA-1 hash of D
		Hsha1(D): <value>
	Form keys Ka and Kb
		Ka: <value>
		Kb: <value>
	Adjust parity bits
		Ka: <value>
		Kb: <value>
DATA - [73, 198, 230, 113, 14, 75, 117, 52]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: <value>
Generate an 8 byte random and a 16 byte random
	RND.IFD: <value>
	RND.Kifd: <value>
Concatenate RND.IFD, RND.ICC and Kifd
	S: <value>
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: <value>
x0: <value>
y0: <value>
x1: <value>
y1: <value>
x2: <value>
y2: <value>
x3: <value>
y3: <value>
x4: <value>
y4: <value>
y: <value>
bkey: <value>
akey: <value>
b: <value>
a: <value>
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: <value>
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: <value>
Error reading tag: sw1 - 69, sw2 - 85 - reason: Conditions of use not satisfied
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 0.)
BAC Failed
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}
```

Getting the error with ReadID demo app as well.

rgex, 2019-11-27:
The german ID Card is not Doc9303 conform, it sucks and is a piece of crap.
To read its content you need a digital certificate from the BSI (costs 100k+).
In addition because the build in chip is so shitty it doesn't work with many android devices.

douglasg, 2019-12-03:
I'm getting a similar error on german id card

```
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 6A, sw2 - 88 - reason: Referenced data not found
BAC Failed
```

AndyQ, 2020-01-08:
Is this still an issue with V1.0.5?

maxxx777, 2020-01-10:
@AndyQ yes, still there

AndyQ, 2020-01-16:
Doesn't look like there is much I can do here so I'm closing this.

If you do manage to get it working with German ID cards and you don't require a special certificate to read it please submit a pull request!

closed 2020-01-16 by AndyQ