AndyQ/NFCPassportReader #29: Wrong length Le: SW2 indicates the exact length

canssamet, 2019-12-02:
Hi @AndyQ 
I get this error in some passports

`Reading  tag - SOD
2019-12-02 12:20:37.508446+0300 TryRead[14053:3034689] [CoreNFC] 00000002 83c1c080
-[NFCTagReaderSession setAlertMessage:]:92  (null)
Error reading tag: sw1 - 6C, sw2 - E7 - reason: Wrong length Le: SW2 indicates the exact length - (exact length :231)`

Tag name changing but error same.
What I should do ?
Are there any temp solution ?

smlu, 2019-12-02:
Can you try setting var [maxDataLengthToRead](https://github.com/AndyQ/NFCPassportReader/blob/master/Sources/NFCPassportReader/TagReader.swift#L180) to `224` and see if it works.

canssamet, 2019-12-02:
> Can you try setting var [maxDataLengthToRead](https://github.com/AndyQ/NFCPassportReader/blob/master/Sources/NFCPassportReader/TagReader.swift#L180) to `224` and see if it works.

Thank you bro but not working. I get same error

smlu, 2019-12-03:
Could you set the log level to `.debug` and post debug log here?

canssamet, 2019-12-04:
> Could you set the log level to `.debug` and post debug log here?

Its working @smlu thank you. NFC didnt read because I changed the wrong line. Now working

AndyQ, 2019-12-10:
I've updated to 1.0.5 - can you please test and let me know if all is well?

AndyQ, 2020-01-08:
Closing as no response - please open new bug if still an issue

closed 2020-01-08 by AndyQ