AndyQ/NFCPassportReader #30: About OpenSSL

puneetmahali, 2019-12-31:
Hi @AndyQ.  I want to use the basic details from the passport like Photo, name, DOB, Address etc. I don't want to use X.509 certificate. So, If I don't want to use x.509 certificate after that also I need to use OpenSSL for fetching the Image or no need.
If yes then how can you I use Open SSL through cocoapods? Because when I install the pod 'OpenSSL-Universal' then also I am unable to Import the OpenSSL Module. Is there any need to make a bridge header too. Can you please explain little more?
Thanks in advance.

puneetmahali, 2020-01-03:
@AndyQ, Can you please tell me the Steps of OpenSSL installation via Pods Setup into the Xcode Project?

AndyQ, 2020-01-08:
I'm using Marcin Krzyzanowski's OpenSSL-Universal pod (https://github.com/krzyzanowskim/OpenSSL) and using the Framework version rather than the static version - if you look in the pod spec you'll see the entry:
  spec.dependency "OpenSSL-Universal/Framework"

The above githubpage should be able to give you more info on how to install it for your use case.

AndyQ, 2020-01-16:
I'm closing this as its not an issue

closed 2020-01-16 by AndyQ