AndyQ/NFCPassportReader #34: How Can I read the Data Group 11?

puneetmahali, 2020-01-21:
Hi @AndyQ 
I want to read the Data Group 11 also. Any approach for that?

puneetmahali, 2020-01-21:
Like DataGroup1 elements stores the all DG1 elements. So, I am getting nil in DG11?
Is there need to define the DG11 tags also like 5F10, 5F11 etc etc. Or can achieve without it?

puneetmahali, 2020-01-21:
Hi @AndyQ 
I am getting an error when I am trying to read the data group 11
Error :
Reading tag - DG11
Error reading tag: sw1 - 6A, sw2 - 82 - reason: File not found
Can you please explain it?

puneetmahali, 2020-01-21:
Thanks @AndyQ  for the quick reply.

Yeah that's true actually my passport read only DG1, DG2, DG 14,SOD & COM. So, now how do I know it's not possible to read Data Group11.
Or, In other words how do we know which data groups are present in passport chip and which is readable through NFC?

Yeah I tested with READ ID and it says DG Read : 1, 3, 14, 2.
Therefore NO DG 11 in my passport but Is there any documentation or something similar which can proof your passport have blah blah data groups.

AndyQ, 2020-01-21:
I really suggest you read the ICAO DOC9303 E-Passport documentation - especially section 10 - https://www.icao.int/publications/Documents/9303_p10_cons_en.pdf

This describes what all the directory items that can be set on a passport are and also says what is mandatory (DG1,2, COM and SOD) and which are optional (pretty much everything else under certain conditions)

The data groups included can be found in the COM entry (which I read into into the dataGroupsPresent field). 

Closing this now as this isn't an issue.

closed 2020-01-21 by AndyQ