AndyQ/NFCPassportReader #35: How can I got the DSC?

puneetmahali, 2020-01-27:
Hi @AndyQ 
After matching the all hashes then how can I verify signature of SOD using the DSC? Can you please confirm the DSC available in chip OR compulsory to receive it from Issuing Authority?

rgex, 2020-01-28:
As far as I know, the DSC is always present on the NFC chip. To verify the chain of trust you need to get the CSCA master list from the ICAO PKD or the BSI or the Schengen Masterlist and this will be enough.

Then you will need verify the chain: CSCA -> DSC -> LDS hash -> Data Group hashes -> Data Group contents and then you are good.

puneetmahali, 2020-01-28:
Hi @rgex,  Thanks for the quick reply.
Great. Is it possible to get the DSC(Document Signing Certificate) with passive authentication to validate the hashes? I already got the Data Groups Hashes and wants DSC without masterlist.

puneetmahali, 2020-01-28:
I mean I will do the checks myself that means need to achieve the DSC's somehow.

rgex, 2020-01-28:
The SOD data group is a PKCS7 file. It contains the DSC. I don't know the current project but using openssl you can deserialize the PKCS7 File and then call p7->d.sign->cert which will return you a STACK_OF(X509) pointer https://github.com/UBIC-repo/core/blob/b83d31c932b40ee908d2f7059d72283d52fd7118/PassportReader/PKCS7/PKCS7Parser.cpp#L76

rgex, 2020-01-28:
Perhaps you can describe us what you want to achieve?
Where do you want to do the verification, on a remote server or on the phone?
For what you are describing you'll need to use either OpenSSL, no way around.

puneetmahali, 2020-01-28:
Yes, I want to achieve the Datagroups data with their hashes and now I want to achieve the DSC certificate so I can validate with those on my own backend. I just need to popup the JSON and PUT into the database for validation checks.
So, In Andy's library how can I got the DSC certificates. I already make the JSON just need to add the DSC also on that JSON.

puneetmahali, 2020-01-29:
Hi @AndyQ , 
`let sod = passport.getDataGroup(.SOD)

  let sodData = Data(sod.body)`

In debugging mode I am getting: DataGroup & pkck7CertificateData = ([UInt8]) 48 values
But I am unable to get the pkck7CertificateData byte data. It comes NULL.

Because I think I am passing .body need to pass pkck7CertificateData but it's not accesible.

And my func look like:
` //MARK: Fetch the Digital Signature Certificate(DSC)
              do {
                guard let sod = passport.getDataGroup(.SOD) else {
                     throw PassiveAuthenticationError.SODMissing("No SOD found" )
                  }
                let sodData = Data(sod.body)
                 let certificateData = try OpenSSLUtils.getX509CertificatesFromPKCS7( pkcs7Der: sodData).first!
                 self.sodDataVal = ["certificateData" : certificateData]
                 print(certificateData)
             }catch {
               print(error)
             }`

So, I need to fetch the pkck7CertificateData UInt8 byte data then I will play around it.

AndyQ, 2020-01-29:
This page may give you some pointers:
http://wiki.yobi.be/wiki/EPassport - specifically the Verifying a passport section.

This is pretty much what I am doing.

AndyQ, 2020-02-07:
Did that help?

AndyQ, 2020-02-12:
Closing as assuming all ok now.

closed 2020-02-12 by AndyQ