AndyQ/NFCPassportReader #36: Objective C Support

jassimkm, 2020-01-29:
How can I use this in objective-c project

danydev, 2020-02-09:
More a question for stackoverflow than for the library itself. Usage should not differ from any other use of Swift code from project with objective-c.
https://medium.com/@anum.amin/swift-pod-library-in-objective-c-project-c6d1c5af997d

alv-ti8m, 2020-10-07:
@danydev @jassimkm  question was actually correct, this framework does not support objective-c. 
You can import, but no methods are marked as @objc. This makes impossible to use on an objective-c project.

Is there any change we can get support for objective-c?

AndyQ, 2023-02-03:
I'm closing this as I have no plans on supporting Objective-C at this time, and there is a reasonable amount of work to implement this.

closed 2023-02-03 by AndyQ