AndyQ/NFCPassportReader #42: Error reading tag: sw1 - 62, sw2 - 82 - reason: End of file/record reached before reading Le bytes

douglasg, 2020-02-11:
With version 1.0.7 it works, with versions 1.0.8 to 1.0.10 I get the following error messages:

```
Starting Basic Access Control (BAC)
BAC Successful
2020-02-11 13:18:09.655779+0100 Identity Reader[3574:1702514] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Reading tag - COM
2020-02-11 13:18:09.701659+0100 Identity Reader[3574:1702514] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Expected response length: 22
Error reading tag: sw1 - 62, sw2 - 82 - reason: End of file/record reached before reading Le bytes
2020-02-11 13:18:09.720678+0100 Identity Reader[3574:1702514] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
ERROR - End of file/record reached before reading Le bytes
Reading tag - COM
Error reading tag: sw1 - 69, sw2 - 88 - reason: SM data objects incorrect
2020-02-11 13:18:09.741806+0100 Identity Reader[3574:1702453] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
ERROR - SM data objects incorrect
Starting Basic Access Control (BAC)
BAC Successful
2020-02-11 13:18:09.812785+0100 Identity Reader[3574:1702514] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Reading tag - COM
2020-02-11 13:18:09.861595+0100 Identity Reader[3574:1702514] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Expected response length: 22
Error reading tag: sw1 - 62, sw2 - 82 - reason: End of file/record reached before reading Le bytes
2020-02-11 13:18:09.882241+0100 Identity Reader[3574:1702498] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
ERROR - End of file/record reached before reading Le bytes
Reading tag - COM
Error reading tag: sw1 - 69, sw2 - 88 - reason: SM data objects incorrect
2020-02-11 13:18:09.904574+0100 Identity Reader[3574:1702514] [CoreNFC] 00000002 834bd800 -[NFCTagReaderSession setAlertMessage:]:92  (null)
ERROR - SM data objects incorrect
Starting Basic Access Control (BAC)
```

this repeats endlessly

Affected version: > 1.0.7
Document: Passport from spain

AndyQ, 2020-02-11:
Could you please try the latest version (currently 1.0.10) - There were some fixes in 1.0.9 that may well have fixed this.

douglasg, 2020-02-11:
> Could you please try the latest version (currently 1.0.10) - There were some fixes in 1.0.9 that may well have fixed this.

Yes, I have also tested 1.0.10, with the same result, and then I gradually went down from that version and the last one that works is 1.0.7

AndyQ, 2020-02-11:
OK, could you please test the following - let me know if that works:

V1.0.10 - TagReader.swift line 294 - change:
    expectedResponseLength : 256
to:
    expectedResponseLength : Int(readAmount)

I have a horrible feeling that your passport doesn't handle the unknown amount of data to be read command (256) but could you confirm that that change fixes the issue?

douglasg, 2020-02-11:
> expectedResponseLength : Int(readAmount)

I can confirm that this fixed the problem

AndyQ, 2020-02-12:
@douglasg I've created a branch - eofreached with a potential fix.  Basically I'm first trying to read an arbitary amount of data (which should fail with your passport).  At this point, I drop down to reading smaller chunks.

So you should see the first SOD read fail, it will then redo BAC and retry again which hopefulyl will work.

Could you please test and let me know?

SaliienkoAleksandr, 2020-02-18:
> > expectedResponseLength : Int(readAmount)
> 
> I can confirm that this fixed the problem

I also had such problem. Int(Amount) fixed it.

SaliienkoAleksandr, 2020-02-18:
Branch - eofreached also works good.

douglasg, 2020-02-18:
@AndyQ Sorry for the late feedback but I have now tested the branch eofreached and it works

AndyQ, 2020-02-18:
@douglasg, @SaliienkoAleksandr - Thanks for testing this!  I'll merge it into master and it will be in the next release

closed 2020-02-18 by AndyQ