AndyQ/NFCPassportReader #44: Implementing Chip Authentication

ChristianNorbertBraun, 2020-02-12:
Hey,
are you thinking of implementing Chip Authentication(CA) besides the Active Authentication(AA)?
Even if CA mostly aims to provide the same verification features as AA, CA seems to be more commonly supported in current passports.

Just for some context, a short summary of CA can be found [here](https://www.bsi.bund.de/EN/Topics/ElectrIDDocuments/SecurityMechanisms/securEAC/securCA/ca_node.html).

rgex, 2020-02-12:
Looks like CA is just another name of AA.

ChristianNorbertBraun, 2020-02-12:
> The following mechanisms to verify the authenticity of the chip are available.
>  1. Active Authentication, as defined in Section 6.1. Support of Active Authentication is indicated by the
> presence of DG15. If available, the terminal MAY read and verify DG15 and perform Active Authentication. 
>  2. Chip Authentication, as defined in Section 6.2. Support of Chip Authentication is indicated by the presence
> of corresponding SecurityInfos in DG14. If available, the terminal MAY read and verify DG14 and
> perform Chip Authentication. 

From the official [icao doc 9303 Part 11](https://www.icao.int/publications/Documents/9303_p11_cons_en.pdf).

Even if they seem to be similar they are actually different.

AndyQ, 2020-02-12:
Yes, I'd like to implement Chip Authentication and additionally PACE but its not straightforward and also testing is an issue (I currently don't have any passports that support this). 

Pull requests welcomed though if you are able to help!

smlu, 2020-03-23:
Just an information for those who would like to implement PACE protocol.
By default [NFCISO7816Tag](https://developer.apple.com/documentation/corenfc/nfciso7816tag) requirers a list of [ISO/IEC 7816](https://cardwerk.com/iso-7816-smart-card-standard/) applets (AIDs). Upon discovery of NFC tag the first found applet from this list is automatically selected (and you have no way of changing this).
This is a problem for PACE protocol becaues it requires reading parameters from file `EF.CardAccess` which lies outside of eMRTD applet (AID: `A0000002471001`) in the master file.
Now, the [ICAO 9303](https://www.icao.int/publications/pages/publication.aspx?docnum=9303) standard does specify command for selecting master file by sending `SELECT` APDU with `P1=0x00`, `P2=0x0C` and empty `data` field (see [part 10](https://www.icao.int/publications/Documents/9303_p10_cons_en.pdf) page 8). But after some testing I found out this command doesn't work on some passports (European passports) and although receiving success (sw=9000) from passport the master file is not selected. 
After a bit of researching standard [ISO/IEC 7816](https://cardwerk.com/iso-7816-smart-card-standard/) I found there is an alternative `SELECT` command for selecting master file. The command doesn't differ much from the command specified in ICAO 9303 doc with only difference that data field is set to: `0x3F00`.  See section 6.11.3 of [ISO/IEC 7816-4](https://cardwerk.com/smart-card-standard-iso7816-4-section-6-basic-interindustry-commands).
By executing above `SELECT` command (with data=`0x3F00`) master file should be selected and you should be able to read `EF.CardAccess` from passport.

rgex, 2020-05-04:
@smlu 

> This is a problem for PACE protocol becaues it requires reading parameters from file EF.CardAccess which lies outside of eMRTD applet (AID: A0000002471001) in the master file.

https://stackoverflow.com/questions/56515747/cannot-read-the-nfc-chip-of-the-epassport-using-ios13/56516232#56516232
May be the link above could solve this issue?

smlu, 2020-05-11:
> 
> 
> https://stackoverflow.com/questions/56515747/cannot-read-the-nfc-chip-of-the-epassport-using-ios13/56516232#56516232
> May be the link above could solve this issue?

@rgex, unfortunately it doesn't work. At least it doesn't work with my European passport.
I tested select identifiers `A0000002471001` in combination with `00000000000000` and select indentifier `00000000000000` alone. None of this identifiers selected the master filer. In the last case passport wasn't detected at all.
My assumption is that my passport doesn't have AID  `00000000000000` for master file, nor should any biometric passport that strictly follows the iCOA 9303 standard.

Btw I haven't found any explanation why Apple choose to go this way and you have to provide AID in advance in order to detect ISO/IEC 7816 tag. And why there is no option to detect general ISO 14443 tag.

rgex, 2020-05-11:
@smlu What kind of passport do you have?
I really need to implement PACE quickly as I get user reports from Korea and Australia that their latest passports don't implement BAC anymore.
Is the work you have done open source?
I am working on this project http://ubic.app/ and any help would be greatly appreciated.

smlu, 2020-05-11:
@rgex I've tested on several European passports pre-2018 and issued in 2018 (after 2018, PACE is mandatory). I had no problem connecting to those passports using BAC protocol. But yeah, I haven't tested any newly issued passport (2019-2020).

FYI I worked on some private passport related projects using Androd/iOS native framework and flutter. Unfortunately, I hadn't developed PACE protocol on any of those project so I can't help you there. I'd share the PACE protocol implementation here. (You probably already know JMRTD library for Android. This library should have the PACE protocol implemented)

Btw thank you for sharing this information here. 
Do you by any chance know what error (SW) do your users get when trying to establish session and BAC is not supported?

rgex, 2020-05-12:
@smlu JMRTD seems quite buggy, I made someone try other open source app that uses it and they got an error (Have you used it?).
I was told that only the ReadID app is working for them.

OpenPACE looks interesting, I currently started writing my own PACE implementation.
I will start with general mapping and ECDH, it appears this variant is the most commonly used.

smlu, 2020-05-13:
> JMRTD seems quite buggy, I made someone try other open source app that uses it and they got an error (Have you used it?).

@rgex Do you maybe know what error did they get?

From my experiences I hadn't had any serious issues with JMRTD. I don't remember though if I used PACE protocol or only BAC to establish SM session.
The only problem that I can recall was calculating hash of serialized EF.DG14 file which didn't match the one stored in the EF.SOD file. The reason behind this issue was that JMRTD doesn't preserve the order of elements in SecurityInfos SET when deserializing. This resulted then in wrong hash being calculated from re-serialized DG14 file.

rpairo, 2020-05-21:
Will there be support for reading the Spanish eid?

bocek, 2020-05-22:
> From the official [icao doc 9303 Part 11](https://www.icao.int/publications/Documents/9303_p11_cons_en.pdf).
>
> Even if they seem to be similar they are actually different.

Hi man

I am having similar problem, my problem is DG15 is present but terminal cannot read the data , what might be the reason ? It seems Active Authentication is not supported because of this i guess.

rgex, 2020-05-24:
@smlu I got it working on Android using JMRTD, although it seems quite slow.
I also saw that you were working on Ethereum projects,and  currently one of the goals is to implement a proof-of-passport as described [here](https://ubic.app/) in Solidity.
[Here is the related discussion](https://github.com/JonahGroendal/x509-forest-of-trust/issues/2)
What is your opinion on this?

smlu, 2020-05-27:
@rgex Have you tried to run the Android app compiled in release mode? Sometimes the release builds reading and writing operations via NFC are faster in release builds due to optimizations (and not logging so much in the background). Though, the spead of transcieve operation usually also depends on eMRTD's chip itself.

As for ethereum, it's a bit out of my league so I really can't comment it.
At the time we were trying to implement solidity contract to verify RSA eMRTD signatures but the project didn't take off so we ended up with one useless Android PoC app and non-functional contract.

himtim, 2021-02-17:
Is CA development still blocking because of no testing documents?

AndyQ, 2021-02-17:
I have one document that supports CA, and have been looking into how JMRTD does it. But not started on implementing it yet (every time I think about it I have a desire to do something easier :)!

himtim, 2021-02-18:
Yes, it seems quite complicated about parsing the infos in DG14:(

AndyQ, 2021-02-25:
I've got Chip Authentication working for DESede based ciphers and it should work for both DH and ECDH keys (DH keys are untested as I haven't a passport that has them)
This should be available in the next release

AndyQ, 2021-02-28:
Does anyone have a Passport that uses AES based keys for Chip Authentication?  I've managed to do the implementation BUT have no idea if it works as no way of testing.
It shouldn't affect how it works though as if it fails it should gracefully handle it and revert back to BAC!

If you do - please let me know!

AndyQ, 2021-03-02:
Release 1.1.1 implements Chip Authentication - both DES and AES based keys are now supported

closed 2021-03-02 by AndyQ