AndyQ/NFCPassportReader #46: BAC Failed on AUT Passport

douglasg, 2020-03-05:
Authentication error with an austrian passport

Debug log (values changed with XXX):
`
2020-03-05 10:44:23.550712+0100 Identity Reader[405:15932] Metal API Validation Enabled
tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x28096e0d0>)
2020-03-05 10:44:32.340169+0100 Identity Reader[405:15932] [CoreNFC] 00000002 8298ff00 -[NFCTagReaderSession setAlertMessage:]:92  (null)
Starting Basic Access Control (BAC)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): XXX
Take the most significant 16 bytes to form the Kseed
	Kseed: XXX
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: XXX
	Calculate the SHA-1 hash of D
		Hsha1(D): XXX
	Form keys Ka and Kb
		Ka: XXX
		Kb: XXX
	Adjust parity bits
		Ka: XXX
		Kb: XXX
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: XXX
	Calculate the SHA-1 hash of D
		Hsha1(D): XXX
	Form keys Ka and Kb
		Ka: XXX
		Kb: XXX
	Adjust parity bits
		Ka: XXX
		Kb: XXX
DATA - [238, 245, 231, 210, 85, 24, 100, 226]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: XXX
Generate an 8 byte random and a 16 byte random
	RND.IFD: XXX
	RND.Kifd: XXX
Concatenate RND.IFD, RND.ICC and Kifd
	S: XXX
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: XXX
x0: XXX
y0: XXX
x1: XXX
y1: XXX
x2: XXX
y2: XXX
x3: XXX
y3: XXX
x4: XXX
y4: XXX
y: XXX
bkey: XXX
akey: XXX
b: XXX
a: XXX
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: XXX
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: XXX
Error reading tag: sw1 - 0x63, sw2 - 0x00
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 20.)
BAC Failed
`

AndyQ, 2020-03-05:
Can i just check thaty if your passport number in the MRZ has any > at the end that you are including those characters in the key?

e.g. if the bottom line on the MRZ looks like:
12345678<8AUT7005233M2507237<<<<<<<<<<<<<<06

In this case the passport number is 12345678 but is padded out with an additonal <.  This needs to be included in the MRZ key used for BAC.
e.g. 12345678<70052332507237 would be the key used.

douglasg, 2020-03-05:
> Can i just check thaty if your passport number in the MRZ has any > at the end that you are including those characters in the key?
>
> e.g. if the bottom line on the MRZ looks like: [...]

@AndyQ , thank you very much, this was the problem.

closed 2020-03-05 by douglasg