AndyQ/NFCPassportReader #50: PACE becomes urgent

rgex, 2020-05-02:
I am getting user reports indicating that BAC is not supported anymore on some passports issued in late 2019 early 2020.

rgex, 2020-05-04:
I found this https://github.com/Governikus/AusweisApp2/tree/23952d6c4b5d8522f0a2526daf3b0f809a18c1b1/src/card/base/pace

It is source code for the German E-ID card, that also uses PACE.
Interestingly they apparently implemented it completely except the 3DES transfer, only AES-12, AES-192 and AES-256
That source code is distributed under the EUPL license so easily reusable.

I will base my work on it and build PACE support using C++ code. I already did that for my other apps, that don't use this library and are open source as well:
 - https://apps.apple.com/gb/app/bondi-passport-reader/id1481019012
 - https://apps.apple.com/us/app/ubic-wallet/id1494645002

kennywen, 2020-06-06:
@rgex do you have any success to enhance the library to support PACE authentication?

AndyQ, 2021-03-05:
I've got an initial version of PACE working - currently only ECDH-GM-AES implemented and tested.

AndyQ, 2021-03-09:
DH AES and 3DES should now also be supported!

Thanks very much to @filom for helping test and bug fix this!

AndyQ, 2021-03-12:
1.1.2 now released which supports PACE GM

closed 2021-03-12 by AndyQ