AndyQ/NFCPassportReader #51: Accessing certificate items

MatthewCochrane-iOS, 2020-05-04:
How do I retrieve certificate items from the X509Wrapper? Any time I try to retrieve them I am receiving a nil value

closed 2020-05-04 by MatthewCochrane-iOS

canssamet, 2020-05-15:
> How do I retrieve certificate items from the X509Wrapper? Any time I try to retrieve them I am receiving a nil value
Hi Matthew . Could you access the certificate items?

MatthewCochrane-iOS, 2020-05-15:
Hi,
Yeah I managed to in the end thank you ☺️

canssamet, 2020-05-15:
@MatthewCochrane-iOS You misunderstood me.Please help :D If you did it can you explain to me ? How can I access Country Signing Certificate  and  Document Signer Certificate  ?

MatthewCochrane-iOS, 2020-05-15:
Ahhhhhh, I get you, I haven't download the correct Masterlist pemfile yet, if I do it some time soon and you haven't got it sorted I will let you know! 🙂