AndyQ/NFCPassportReader #55: ld: warning: Could not find auto-linked library 'swiftObjectiveC'

tricpony, 2020-05-21:
After flip flopping bitcode build settings all over I finally end up with the link error in the subject.  I can not find the proper build setting to resolve this.  I know it has something to do with OpenSSL having bitcode off, but beyond that, suggestions?  I am trying to deliver the functionality of your pod through my own framework.  It all compiles fine while working directly with the framework, not so much when I try to link that framework to my app.  Tossing the framework is not an option and I would rather not turn it into a pod.

AndyQ, 2020-11-17:
This may be fixed by moving over to the newly release SPM version of OpenSSL - will be testing that out.

AndyQ, 2021-03-15:
Closing as I think this should be fixed with SPM support now.  Please open a new issue if required.

closed 2021-03-15 by AndyQ