AndyQ/NFCPassportReader #56: Cannot read DG15

bocek, 2020-05-22:
Hi , i am having problem with reading DG15 data, there is no problem with DG1, DG2 etc. 
In the code below ,  **OpenSSLUtils.readECPublicKey( data:body )**  cannot read data here, returns empty. And throws unabletoread ec public key error. What might be 
 

 public class DataGroup15 : DataGroup {
    
    public private(set) var rsaPublicKey : OpaquePointer?
    public private(set) var ecdsaPublicKey : UnsafeMutablePointer<EVP_PKEY>?
    
    deinit {
        EVP_PKEY_free(ecdsaPublicKey);
    }

    required init( _ data : [UInt8] ) throws {
        try super.init(data)
        datagroupType = .DG15
    }
    
    
    override func parse(_ data: [UInt8]) throws {
        
        // the public key can either be in EC (elliptic curve) or RSA format
        // Try ec
        
        if let key = try OpenSSLUtils.readECPublicKey( data:body ) {
            // NOTE We are responsible for freeing the key!
            ecdsaPublicKey = key
        }
    }
}

himtim, 2021-02-04:
I am also facing same issue, while reading a passport only contains DG15 without DG14 (which is containing Security Options), it will fail in generating the eckey in OpenSSLUtils.swift. 
Not sure the reason, but it is success on the passport which DG15 and DG14 both contains.

Is current default public key algorithm is ECDSA? if fail, then use RSA?
if yes, it should be not related to default algorithm as I expect to use ECDSA for the passport I am testing.
Thank you.

        guard let eckey = d2i_EC_PUBKEY_bio(inf, nil) else { throw OpenSSLError.UnableToReadECPublicKey("Failed to load") }

AndyQ, 2021-02-05:
If anyone is happy to send me either their read passport data or just the DG15 part I'm happy to take a look - the only passport I have available with DG15 contains a ECDSA key.

Please  email me if you can help (email in profile) and I'll send you details of how to capture this.

Thanks!

himtim, 2021-02-05:
Thank you for your reply!

Checked my passport DG15 is in RSA format, so that there is an error in reading public key.

May I know do you have a plan of developing RSA key verification?

I will also try to work on it but I am not familiar with it.

Thank you!

AndyQ, 2021-02-05:
Absolutely - I've been working on a few changes around passive authentication and have RSA verification working there, so hopefully this can also be used here.  However as mentioned, I'm hampered on this by not having access to any passports with this format!

Please drop me an email if you are willing to share the DG15 and a sample auth challenge.

himtim, 2021-02-06:
Do you want the DG15 part is the "body" from this code? and where can I find a sample auth challenge?
I am not familiar with these....thank you!

        // the public key can either be in EC (elliptic curve) or RSA format
        // Try ec
        if let key = try OpenSSLUtils.readECPublicKey( data:body ) {

AndyQ, 2021-02-06:
If you could add the following to the top of NFCPassportModel : verifyActiveAuthentication() :
`    public func verifyActiveAuthentication( challenge: [UInt8], signature: [UInt8] ) {
        
        Log.debug( "Active Auth:")
        Log.debug( "Challenge - \(binToHexRep(challenge))" )
        Log.debug( "Signature - \(binToHexRep(signature))" )
`
This is a randomly generated challenge signed by the passport internal chip - I'll add this into the next version so it is captured too).

You can then use the sample app to export the passport to JSON. Please remove the other DGs from this file but keep the DG15 one.

Also, if this is OK, please leave in the SOD as it would be useful to test my Passive Auth changes! (it doesn't contain any personal info only hashes of the other DGs, the document signing certificate and some signatures).

Then send me the JSON file and the above debug output 

Thanks very much!

himtim, 2021-02-08:
I dropped an email with the DG15 and SOD info to you.

I commented some code in OpenSSLUtil.readECPublicKey() to make parsing DG15 can pass as 

    static func readECPublicKey( data : [UInt8] ) throws -> UnsafeMutablePointer<EVP_PKEY>? {
    
        guard let inf = BIO_new(BIO_s_mem()) else { throw OpenSSLError.UnableToReadECPublicKey("Unable to allocate output buffer") }
        defer { BIO_free(inf) }

        let _ = data.withUnsafeBytes { (ptr) in
            BIO_write(inf, ptr.baseAddress?.assumingMemoryBound(to: UInt8.self), Int32(data.count))
        }
        
        // guard let eckey = d2i_EC_PUBKEY_bio(inf, nil) else { throw OpenSSLError.UnableToReadECPublicKey("Failed to load") }
        // defer{ EC_KEY_free(eckey) }

        guard let outf = BIO_new(BIO_s_mem()) else { throw OpenSSLError.UnableToReadECPublicKey("Unable to allocate output buffer") }
        
        defer { BIO_free(outf) }
        //let _ = PEM_write_bio_EC_PUBKEY(outf, eckey);
        let pemKey = PEM_read_bio_PUBKEY(outf, nil, nil, nil)
        return pemKey
    }

I guess it should be not affect the content of DG15.

However I can't do the sample auth challenge because it is blocked by tagReader?.doInternalAuthentication.
Could you help on this?
` 
        let challenge = generateRandomUInt8Array(8)
        self.tagReader?.doInternalAuthentication(challenge: challenge, completed: { (response, err) in
            if let response = response {
                self.passport.verifyActiveAuthentication( challenge:challenge, signature:response.data )
            }

            completed()
        })
` 

Thank you very much!

himtim, 2021-02-08:
Just found the the active auth in the debugger.
I replied my email of these info, thank you!

AndyQ, 2021-02-08:
Thanks very much! Really appreciated.

AndyQ, 2021-02-09:
This should be fixed in 1.0.17. That adds support for Active Authentication with RSA based signature and keys

closed 2021-02-09 by AndyQ