AndyQ/NFCPassportReader #57: App got crashed while fetching DG1 from french passport

RaknasAvis, 2020-06-04:
while reading the datagroups it got crashed

 tag.sendCommand(apdu: toSend) { [unowned self] (data, sw1, sw2, error) in
            if let error = error {
                Log.error( "Error reading tag - \(error.localizedDescription)" )
                completed( nil, TagError.ResponseError( error.localizedDescription, sw1, sw2 ) )
            } else {
                var rep = ResponseAPDU(data: [UInt8](data), sw1: sw1, sw2: sw2)
                
                if let sm = self.secureMessaging {
                    do {
                        rep = try sm.unprotect(rapdu:rep)

I tried to debug, the "data" was empty and also there is no error(nil). So while "unprotect" response it got crashed.

        // Mandatory, only absent if SM error occurs
        do99 = [UInt8](rapduBin[offset..<offset+4])
This the exact line, it got crashed

AndyQ, 2021-02-12:
Is this still happening?  I've been testing with some French passports and they are working ok.

advatar, 2021-02-25:
Some French passports have DataGroup13 which is marked as NotImplemented generating an exception.

AndyQ, 2021-02-25:
Aha thanks.  A workaround initially is remove the throw line from DataGroup. swift parse method:
e.g.:

    func parse( _ data:[UInt8] ) throws {
    }

I'll fix this in the next release BUT if you are willing, could you send me the DataGroup13 data (use the Export Passport feature of the sample app) and I can look at implementing parsing that! 

If you could also, please include the SOD, COM, DG14 and DG15 groups too - but not the others. These contain non-sensitive information (public keys and signatures) which are useful for me to make sure I don't break.

Thanks

AndyQ, 2021-03-02:
This should be fixed in release 1.1.1

closed 2021-03-02 by AndyQ