AndyQ/NFCPassportReader #59: Error while unpacking certs from LDIF

radimmotycka, 2020-06-16:
Hey Andy thanks for nice library. By any chance would you advice with issue below?
I downloaded The latest collection of CSCA Master Lists from https://download.pkd.icao.int/download as you adviced but getting error while unpacking certs.

[image]

ekam123, 2020-06-22:
I had the exact same error (using openssl 1.1.1g on mac) with the latest list - icaopkd-002-ml-000147.ldif. I commented out line 88 in extract.py (raise Exception( "Verification of Masterlist data failed" )) so it wouldn't crash. There were a total of 10 Masterlists that were looped over and the error showed up only in Masterlist 7.  I added the generated pem file to xcode and it worked for me, although I just needed Canada. I wonder what countries may be left out by skipping the 7th loop but since many of them have duplicates, is there a chance none get left out?

ruudk, 2020-07-15:
Same issue here with `icaopkd-002-ml-000150.ldif`:
```
Verifying and extracting MasterList 8
[Verification failure
4327046592:error:04091068:rsa routines:int_rsa_verify:bad signature:crypto/rsa/rsa_sign.c:220:
4327046592:error:2E09809E:CMS routines:CMS_SignerInfo_verify:verification failure:crypto/cms/cms_sd.c:767:
]
```

ruudk, 2020-07-15:
I found out it's related to the version of OpenSSL on MacOS (installed via Homebrew).

This fixed it for me:

```
# Install older version of OpenSSL (separately)
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/8b9d6d688f483a0f33fcfc93d433de501b9c3513/Formula/openssl.rb

# Add the location to your PATH, I use Fish so it's:
set -g fish_user_paths "/usr/local/opt/openssl/bin"

# Then run the command
python3 extract.py icaopkd-002-ml-000150.ldif

# Victory!
```

AndyQ, 2020-07-17:
Yes the version of OpenSSL that comes with the Mac doesn't support the CMS commands required.
Its mentioned in the Readme for the scripts but I'll add the above information to make it clearer!

Thanks!

closed 2020-07-17 by AndyQ