AndyQ/NFCPassportReader #60: "Unknown error - sw1: 0x63, sw2 - 0xCF ", 99, 207

brqgoo, 2020-07-08:
With version 1.0.11 it works well with pretty much all passports. However, I get the following error message when tried with a Chinese passport:
`Sorry, there was a problem reading the passport. ("Unknown error - sw1: 0x63, sw2 - 0xCF ", 99, 207)`

[image]

AndyQ, 2020-12-15:
Not sure what this error means - I think it could possibly be a verify failed - but exactly what - I'm not sure.
Do you have any more detail on what was happening?

AndyQ, 2021-02-12:
This should be fixed (I hope). If you are still having issues please re-open.

closed 2021-02-12 by AndyQ

advatar, 2021-02-25:
@AndyQ What did you change to fix this issue? I am getting it for Argentinian passports and my code is a bit out of sync with master so I need to merge manually.

AndyQ, 2021-02-25:
I think this was either the change to drop the data read amount down to 0xA0 (160), or possibly the padding around the MRZ Passport number (if its less than 9 chars, it needs to be padded with <)

advatar, 2021-02-25:
Thanks, I am already doing the padding. Will try 0xa0 as the starting point.

advatar, 2021-02-25:
@brqgoo Is this working for you now?

advatar, 2021-02-25:
@AndyQ thanks for the project. I am using much of it for https://svipe.com

brqgoo, 2021-02-25:
> @brqgoo Is this working for you now?

Unfortunately, I don't have access to a Chinese passport atm. I can give an update when I have a chance to test it.

advatar, 2021-02-25:
Hmm, I changed to 160 and now have a problem with invalid MRZ

advatar, 2021-03-01:
I changed to 160 and still get the same problem

[image]

Any ideas what to do next?

advatar, 2021-03-04:
Turns out that https://github.com/Mattijah/QKMRZParser converts dates to Date with a particular TZ so when computing the MRZ for BAC dates are off by one if you are west of GMT...