AndyQ/NFCPassportReader #64: error 10 at 2 depth lookup:certificate has expired

ruudk, 2020-07-15:
I'm testing out the Example application with 2 EU passports from The Netherlands. 

One is from 2017 and is marked as ✅ Genuine. 
One is from 2015 and is marked as :x: Not Genuine

For this one I see the following error in the console:
```
error 10 at 2 depth lookup: certificate has expired
```

I wonder what the problem is? It says the `certificate has expired`. Which certificate is this? The `masterList.pem` file? Or the certificate on the chip?

When I use [RealID](https://apps.apple.com/us/app/readid-nfc/id1463949991) app on iOS it works perfectly. 

I've constructed the `masterList.pem` file from the latest `icaopkd-002-ml-000150.ldif` file.

ruudk, 2020-07-15:
With `RealID` I see that it uses `PACE` to access the 2017 passport and BAC for 2015.

AndyQ, 2020-10-01:
I would assume that this is the certificate in the Masterlist - have you tried with an updated list?

Also, I assume you mean that 2015/2017bis the dates the passports were issued and not expired?

AndyQ, 2021-03-15:
I'm closing this as no update provided - if its still an issue please re-raise.

closed 2021-03-15 by AndyQ