AndyQ/NFCPassportReader #65: Not detecting tag for US passport

juliana88, 2020-07-20:
Hi, 

I've implemented your code inside my project. Upon testing on a US passport, first the **tagReaderSessionDidBecomeActive(_ session: NFCTagReaderSession)** get called. After a while, it called **tagReaderSession(_ session: NFCTagReaderSession, didInvalidateWithError error: Error)** with error of "Session timeout" without calling the **tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag])** at all.  I've already included "ISO7816 application identifiers for NFC Tag Reader Session" in my project Info.plist. Hope you can help me in the direction to solve this problem. Thank you

brqgoo, 2020-07-23:
Hi Juliana, can you try with ReadID app this time? If ReadID not works too, It's bc some US passports are somehow not responding to NFC reading session even though they have a cover bearing the ICAO "chip inside" logo. EU passports are OK, but some US passports are problematic for some reason. IMO

juliana88, 2020-07-28:
The issue happens cause the tester is not holding the iDevice and passport still during the NFC detection phase.

closed 2020-07-28 by juliana88

phisakel, 2021-01-09:
You can try adding tagRemovalDetect from here: https://blog.st.com/wp-content/uploads/DemoEHViewController.swift