AndyQ/NFCPassportReader #70: Is chip authentication useful?

wellbranding, 2020-09-08:
Hello, I am coming here from the NFC reader repository in Android. A guy suggested this repository in [this thread](https://github.com/tananaev/passport-reader/pull/19). I looked at this repository and found that you are using active identification instead of the chip authentication. Why have you decided to omit chip authentication? Is it less valid than active authentication?
We have discussed more about chip authentication [here](https://github.com/tananaev/passport-reader/pull/19). If chip authentication is indeed more secure than active authentication I would be grateful to contribute :)

TWilb, 2020-09-08:
Like RomainL972 wrote in the mentioned thread:
Both are used to protect documents against cloning. Active Authentication (AA) is older and mainly used in documents without Extended Access Control (EAC).
Chip Authentication (CA) was introduced together with Terminal Authentication (TA) and is part of EAC.

In the field you can find passports and ID cards with none, just one or even both protocols available. So ideally both is supported for checking different documents and if both is available on a document I would prefer using CA.

I think AA was done first because of its simpler approach.

wellbranding, 2020-09-08:
Thanks for the quick response. Are you already in progress implementing CA?

wellbranding, 2020-09-15:
Hello, sorry for double-checking, but are you planning to implement CA?

TWilb, 2020-09-18:
I am not one of the developers, so I cannot say if this is planned.

AndyQ, 2020-09-21:
I'd like to implement Chip Authentication - its on the backlog.  The main issue preventing me is that I don't currently have access to a passport that supports this which makes developing and testing it a little tricky!

typelogic, 2021-02-05:
I do not think that the explanation can be found nor deduced by just staring at the source codes. Attached please find [ICAO informative notes](https://imgur.com/vhyPSMR.png) with regards to an accurate description of **AA** versus **CA**.

AndyQ, 2021-02-25:
I've got Chip Authentication working for DESede based ciphers and it should work for both DH and ECDH keys (DH keys are untested as I haven't a passport that has them)
This should be available in the next release

AndyQ, 2021-03-02:
Release 1.1.1 implements ChipAuthentication

closed 2021-03-02 by AndyQ