AndyQ/NFCPassportReader #71: Get UID And Serial Number of the NFC Tag iOS

shamdass, 2020-09-15:
I'm developing an application in which I want to get the NFC serial number and Identifier of a NFC tag. When I press a button, I want the application to find the UID and serial number of the NFC card. The only problem is that I have no idea how to get the UID and and serial number if that's possible on iOS. I m trying to get NFC identifier but i don't know how to get serial number of NFC Tag. 

Code:- 

    class ScanNFCTag: NFCTagReaderSessionDelegate {

    import CoreNFC

    var session: NFCTagReaderSession?

    func scanNFCTag(sender: UIButton!) {
         self.session = NFCTagReaderSession(pollingOption: .iso14443, delegate: self)
         self.session?.alertMessage = "Hold Your Phone Near the NFC Tag"
         self.session?.begin()
    }

    func tagReaderSessionDidBecomeActive(_ session: NFCTagReaderSession) {
        print("Session Begin!")
    }
    
    func tagReaderSession(_ session: NFCTagReaderSession, didInvalidateWithError error: Error) {
        print(error.localizedDescription)
    }
    
    func tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag]) {
        if tags.count > 1{
            session.alertMessage = "More Than One Tag Detected, Please try again"
            session.invalidate()
        }
        
        let tag = tags.first!
        session.connect(to: tag) { (error) in
            if nil != error{
                session.invalidate(errorMessage: error?.localizedDescription ?? "")
            }
            if case let .miFare(sTag) = tag{
                let UID = sTag.identifier.map{ String(format:"%.2hhx", $0)}.joined()
                session.alertMessage = "Successfully linked tag to location"
                session.invalidate()
                DispatchQueue.main.async {
                    print("NFC Identifer = ", "\(UID)")
                   }
               }
           }
       }
    }

So, Here is the my code please let me know if i m doing wrong please correct me. Can someone please explain to me how to get serial number and UID of NFC tag,

Any help would be greatly appreciated.

Thanks in advance.

AndyQ, 2020-09-21:
Sorry, I'm closing this as isn't an issue for this project

closed 2020-09-21 by AndyQ

shamdass, 2020-09-21:
@AndyQ @ruudk @WooD1k @danydev  Yeah, I know this is not an issue, but if you know this how to get please help me, I really thankful for you