AndyQ/NFCPassportReader #72: XCode 12.0 compile error

Shamsiddin, 2020-09-19:
Tools:
XCode version: 12.0 (12A7209)
Simulator: iPhone 8 (iOS 13.4)
Project language: Swift 5
MacOS version: 10.15.4

Everything was working well in a XCode 11.5
With XCode 12.0 giving an error as shown in a screenShot:

[image]

Looking forward to the new version of the Lib!

AndyQ, 2020-09-21:
I've got a fix for this - I haven't yet been able to test it (as I currently don't have access to my test passport that supports Active Authentication)

If you replace lines 321-323 in the above with :
        let sigData = signature
        sigData.withUnsafeBufferPointer { (unsafeBufPtr) in
            let unsafePointer = unsafeBufPtr.baseAddress!
            BN_bin2bn(unsafePointer, 32, ecsig?.pointee.r)
            BN_bin2bn(unsafePointer + 32, 32, ecsig?.pointee.s)
        }

That will fix the compile issue and should work (but I can't confirm)

AndyQ, 2020-09-21:
Tested an 1.0.12 pushed - please test and let me know if this resolves your issue

Punkers, 2020-09-24:
I was having the same issue as @Shamsiddin, so I updated the library to 1.0.12.
The compilation error is gone, but now the app crashes on startup with the following error

[image]

I experienced it both on your example app and on a test app, both on device and simulator.

XCode version: 12.0 (12A7209)
Simulator: iPhone 11 (iOS 14)
Device: iPhone 11 (iOS 14)
Project language: Swift 5
MacOS version: 10.15.6

Shamsiddin, 2020-09-30:
@AndyQ sorry for late response!
It was worked perfectly. Thanks!

closed 2020-10-01 by AndyQ