AndyQ/NFCPassportReader #73: BAC Failed when reading swedish passports

LinusBF, 2020-09-24:
Hi!

I was trying the example application and tried to read my swedish passport.
The MRZ key seems to be generated correctly but I get this output when I try to scan it:

```
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 0x63, sw2 - 0x00
BAC Failed
```

Are there some limitations in regards to what passports are supported?
If so, do you have any pointers on how to get that working?

Info:
XCode - 12.0
Iphone - 11
iOS - 13.6.1

jkufver, 2020-09-29:
Have you added the extra padding "<"  to the end of document number? It must be 9 digits long.

jkufver, 2020-09-29:
Reported the same a while ago: https://github.com/AndyQ/NFCPassportReader/issues/47

LinusBF, 2020-09-29:
Yes, that solved it!
Thank you very much!

closed 2020-09-29 by LinusBF