AndyQ/NFCPassportReader #74: runtime error on iPhone 7 - iOS 14

ssinan, 2020-09-30:
Hello,

Example project gets runtime error on iPhone 7 (iOS 14). It works correctly on iPhone 8 Plus (iOS 14) and iPhone 11 (iOS 14) (I tested).

Error details
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 0x65, sw2 - 0x81
reason: Memory failure
BAC Failed

I thought my iPhone 7 might have some problem so I downloaded another app on AppStore (ReadID NFC). It worked.

sonmezvolkan, 2020-10-23:
Hi 
Same problem different error
Your library has worked so far on iPhone 7 ios 13, but does not work on ios 14 after os update.
Could that be anything related to operating system update since it was working ios 13 ?
My error details
Starting Basic Access Control (BAC)
Error reading tag: sw1 - 0x69, sw2 - 0x82
reason: Security status not satisfied
BAC Failed

ebrienza, 2020-11-02:
Hi,

Same problem.
Same eID, same smartphone, same source-code.
No problem with iOS 13 but after iOS14 and iOS14.1 update I cannot initiate s stable communication between card and smartphone.

wilk-polarny, 2020-11-04:
Hi, we're having the exact same issue where scanning works on newer iPhones (iPhone 11), but not on iPhone 7 models.

The app itself logs:

```
2020-11-04 16:55:51.343150+0100 MPK.Abfragen[356:7856] [CoreNFC] 00000002 817a9780 -[NFCTagReaderSession _connectTag:error:]:560  Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x283787ab0 {Error Domain=nfcd Code=29 "Tag Error" UserInfo={NSLocalizedDescription=Tag Error}}}
```

I have attached logs of a successful scan and an unsuccessful scan:

[FAILED - iPhone7 - nfclogs-i7.log](https://github.com/AndyQ/NFCPassportReader/files/5489389/nfclogs-i7.log)
[SUCCESSFUL - iPhone 11 -  nfclogs-i11.log](https://github.com/AndyQ/NFCPassportReader/files/5489391/nfclogs-i11.log)

The output of nfcd differs, but I guess mostly due to the fact that they use different rfid hardware?

wilk-polarny, 2020-11-05:
@ssinan , @sonmezvolkan , @ebrienza - did anyone of you escalate that issue to Apple? Cause it looks like an iOS issue, as other solutions like ReadID exhibit the same behaviour on iPhone 7 devices.

AndyQ, 2020-11-05:
Interesting - Unfortunately I currently don't have an iPhone 7 available to test this against (and won't for at least a month or two) but it does look like an underlying iOS issue.

Its working on my iPhone 11 Pro and an Phone 11 I've got access to.

The above logs seem to show it just losing connection to the NFC Chip - this happens sometimes if the phone isn't quite oriented correctly or slightly too far away.

The BAC Error 0x69 0x82 - usually means that we can't create the secure session - possibly the key is incorrect (unlikely but possible) but that is a more generic error that is returned by the chip when it thinks there is unauthorised access.

The 0x65, 0x81 Memory error - I have no idea at all - never seen this!

ebrienza, 2020-11-05:
@wilk-polarny  I sent a bug report to Apple with Feedback Assistant. Maybe you should do the same.

wilk-polarny, 2020-11-05:
@AndyQ @ebrienza , thank you. We'll likely try to raise a TSI with Apple as using the Feedback tool usually leads to no results. Still have some issues open since 2014 where they only ask to update to the latest macOS and confirm whether the issue is still there ^^

ebrienza, 2020-11-05:
@wilk-polarny done.

sonmezvolkan, 2020-11-05:
@AndyQ thank you.
@wilk-polarny @ebrienza Have you tried another application ? like ReadID NFC as @ssinan said. I have tried and it worked correctly.

wilk-polarny, 2020-11-05:
@sonmezvolkan , tried with ReadID on an iPhone 7 and it didn't work, wheras it does on an iPhone 11. Same goes for this library and other solutions. Could this also be related to specific passports?

ebrienza, 2020-11-05:
No @wilk-polarny is not related to your specific passport. I use NFCTagReaderSession iso14443 on some product without this library for interact with Italian eID and we have the same problem but also with iPhone 7/7plus smartphones. User reviews, logs, my tests and other my colleagues confirms this problem.

wilk-polarny, 2020-11-06:
Looks like this issue was fixed in the latest iOS Release iOS 14.2! (at least here)
Can you guys also confirm?

sonmezvolkan, 2020-11-07:
Finally. i could try and it worked correctly. 
@wilk-polarny thank you for information.

Thank you all. 

i think we can close this issue

ebrienza, 2020-11-07:
It works!

AndyQ, 2020-11-08:
Closing this now as it looks like Apple have fixed this issue

closed 2020-11-08 by AndyQ

wellbranding, 2022-07-07:
@ebrienza Hello, we as well facing issues with the Italian passport. Can you check if yours is this one?
[https://www.consilium.europa.eu/prado/en/ITA-AO-02003/index.htm](https://www.consilium.europa.eu/prado/en/ITA-AO-02003/index.html)
Have you resolved the issue?