AndyQ/NFCPassportReader #81: Nothing happens / Tag Error 2

g522811, 2020-11-20:
I've tested both the sample app and my own implementation on an iPhone XR and iPhone 8. The sample app always gives Tag Error 2 and my own implementation always gets a timeout error. I've tried front back and insides of a U.S. passport against the back surface of the iPhones.

Any things I can try?

g522811, 2020-11-24:
Behavior is similar to #65 and #26 
- I'm laying the device completely still on top of the passport (opened to the last page where the chip is embedded)
- Testing on a United States passport that successfully scans with Android NFC devices
- Testing on an iPhone XR and iPhone 8 both running 13.5.1
- .entitlements contains:
`<key>com.apple.developer.nfc.readersession.formats</key>`
`<array>`
`<string>NDEF</string>`
`<string>TAG</string>`
`</array>`
- `Info.plist` contains:
`<key>NFCReaderUsageDescription</key>`	
`<string>This application uses NFC for ...</string>`
`<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>`
`<array>`
`<string>A0000002472001</string>`
`<string>A0000002471001</string>`
`<string>00000000000000</string>`
`</array>`
- Signing & Capabilities includes _Near Field Communication Tag Reading_
- Both the sample app and my implementation fail to work

@AndyQ any ideas?

g522811, 2020-11-24:
Solved by providing valid AID

closed 2020-11-24 by g522811

atatlisu, 2025-11-24:
@g522811 hello, i receive InvalidMRZKey error on usa passports while other passports works fine and i am sure the mrz key is correct i try pace first if it fails i try bac any idea ? and how did you solve it by providing valid AID ? what is valid AID for usa passports and where did you add it ? thanks in advance. best regards

> Solved by providing valid AID