AndyQ/NFCPassportReader #86: DO99 Fatal error: Array index is out of range

andycg0327, 2020-12-02:
SecureMessaging.swift
Line 117: Fatal error: Array index is out of range

Only happen at iPhone12 (14.2.1)

AndyQ, 2020-12-15:
Could you provide some more logs - you can change the log level passed into the PassportReader.

Thanks

andycg0327, 2020-12-25:
Starting Basic Access Control (BAC)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): 357423084252D8C51BD9676608E98A87485CA377
Take the most significant 16 bytes to form the Kseed
	Kseed: 357423084252D8C51BD9676608E98A87
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: 357423084252D8C51BD9676608E98A8700000001
	Calculate the SHA-1 hash of D
		Hsha1(D): CA1D2A2AC47883444B5DA0271084F75000943300
	Form keys Ka and Kb
		Ka: CA1D2A2AC4788344
		Kb: 4B5DA0271084F750
	Adjust parity bits
		Ka: CB1C2A2AC4798345
		Kb: 4A5DA1261085F751
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: 357423084252D8C51BD9676608E98A8700000002
	Calculate the SHA-1 hash of D
		Hsha1(D): 2772972C66000964E89A2F7D71CEEF7FC4E0A348
	Form keys Ka and Kb
		Ka: 2772972C66000964
		Kb: E89A2F7D71CEEF7F
	Adjust parity bits
		Ka: 2673972C67010864
		Kb: E99B2F7C70CEEF7F
DATA - [182, 66, 184, 78, 108, 106, 1, 117]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: B642B84E6C6A0175
Generate an 8 byte random and a 16 byte random
	RND.IFD: 95CCB86F736028B3
	RND.Kifd: 1D70ED9248CE477817E6DF8F85ABDE05
Concatenate RND.IFD, RND.ICC and Kifd
	S: 95CCB86F736028B3B642B84E6C6A01751D70ED9248CE477817E6DF8F85ABDE05
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: 79B1A52CE212BD898DDAF3965454BAD6DFCAD63E32DE01A4D84093A236E781A5
x0: 79B1A52CE212BD89
y0: 4A194433E3FCB23F
x1: 8DDAF3965454BAD6
y1: 348FE6FA31AB2728
x2: DFCAD63E32DE01A4
y2: E1ADEC574426DF0E
x3: D84093A236E781A5
y3: 4DE27666DE84729B
x4: 8000000000000000
y4: C9DA4A7E55D96E88
y: C9DA4A7E55D96E88
bkey: E99B2F7C70CEEF7F
akey: 2673972C67010864
b: 85B102635071C1E9
a: E9E449DC9F07EC45
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: E9E449DC9F07EC45
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: 79B1A52CE212BD898DDAF3965454BAD6DFCAD63E32DE01A4D84093A236E781A5E9E449DC9F07EC45
DATA - [213, 210, 180, 153, 190, 169, 243, 117, 236, 131, 73, 190, 166, 198, 177, 31, 16, 93, 219, 48, 42, 33, 105, 116, 196, 187, 199, 56, 232, 191, 170, 29, 87, 40, 186, 183, 174, 176, 151, 94]
Decrypt and verify received data and compare received RND.IFD with generated RND.IFD 2673972C67010864E99B2F7C70CEEF7F
Calculate XOR of Kifd and Kicc
	Kseed: 1FD252588C4E7F006E02D662D405F988
	Concatenate Kseed and c
		D: 1FD252588C4E7F006E02D662D405F98800000001
	Calculate the SHA-1 hash of D
		Hsha1(D): C9605E2B26627888E939C8D79729322A9B498E08
	Form keys Ka and Kb
		Ka: C9605E2B26627888
		Kb: E939C8D79729322A
	Adjust parity bits
		Ka: C8615E2A26627989
[... 4018 lines cut ...]
y0: FB7D0358E4AC0A6B
x1: 0CB0112980000000
y1: 3105C60D6AD38C45
x2: 9701008000000000
y2: 75F13B6B03E330A0
y: 75F13B6B03E330A0
bkey: E638769E34981349
akey: FD4AC70407EA4954
b: 7CA280B74CC07D47
a: 243600867DE02D3A
	Compute MAC over N with KSmac
		CC: 243600867DE02D3A
Build DO'8E
	DO8E: 8E08243600867DE02D3A
Construct and send protected APDU
	ProtectedAPDU: 0CB011290D9701008E08243600867DE02D3A00
[SM] [0x0C, 0xB0, 0x11, 0x29, 0x0D, 0x97, 0x01, 0x00, 0x8E, 0x08, 0x24, 0x36, 0x00, 0x86, 0x7D, 0xE0, 0x2D, 0x3A, 0x00]
Receive response APDU of MRTD's chip
	RAPDU: 8781E901D8C440269B36A611DEC2CB4460B4F1839D0F3825F60B1E737DD240DD4D9DFD35E3258F5D42A58B0EBE09C1435FA3BF0B15B49D951AE38029BF56B11ED3496F8110A09DEA60558C12F6EBBA491D6132EF15DAA3F9F081B8C7FD3D74C2ADB166A698754BA416CE82E7E31149A4D89E3B3C283FA9C68D44B57DF6D1963539D9FA3A369FCCDC0C1A3DDAD1F72439A3660F57DEFC4E9AD64A1F0BEE34F11817127366A6CCDE24BBA616B300D2087A037A5B2177B6474964CDB2727EFBBC972AA6732500356524E35133B3FF61C666F8561FB9E9381423D2E09BD0D3E9776DB8628B47F85311D37903D5A4990290008E0809DA20B75A9AACD59000
Verify RAPDU CC by computing MAC of DO'87 DO'99
		SSC: 6C6A017573602902
	Increment SSC with 1
		SSC: 6C6A017573602903
	Concatenate SSC and DO'87 DO'99 and add padding
		K: 6C6A0175736029038781E901D8C440269B36A611DEC2CB4460B4F1839D0F3825F60B1E737DD240DD4D9DFD35E3258F5D42A58B0EBE09C1435FA3BF0B15B49D951AE38029BF56B11ED3496F8110A09DEA60558C12F6EBBA491D6132EF15DAA3F9F081B8C7FD3D74C2ADB166A698754BA416CE82E7E31149A4D89E3B3C283FA9C68D44B57DF6D1963539D9FA3A369FCCDC0C1A3DDAD1F72439A3660F57DEFC4E9AD64A1F0BEE34F11817127366A6CCDE24BBA616B300D2087A037A5B2177B6474964C

AndyQ, 2021-02-25:
This should be fixed in the current version of the app.  The problem was down to the amount of data being read.  Dropping this down to a more supported amount (rather than trying to read until the passport said it had finished which isn't supported by all passports apparently) should fix this.  

If you are still having issues please re-open

closed 2021-02-25 by AndyQ

andycg0327, 2023-01-12:
Thank you for the fix. 
It been a while since the last check, unfortunately, today we tried in iPhone (XR, SE, 12, 13), the same issue still happen and just only in iPhone12...

Can you please help for this problem again? Thank you so much.