AndyQ/NFCPassportReader #89: maxDataLengthToRead wrong value

nikolindima, 2020-12-20:
Hello. need to change maxDataLengthToRead to 255. 256 not working with some passports

public class TagReader {
    var maxDataLengthToRead : Int = 255

smlu, 2020-12-20:
Hey @nikolindima, could you tell us what error you get when trying to read binary data with `maxDataLengthToRead=256` and which countries issued those failed passports?
The constant `256` is not an actually size of returned data chunk but it tells passport chip to return arbitrary long data (it should be encoded as `0x00` in command APDU). So I'm guessing the failed passports don't support arbitrary data read because errors *"Wrong length"* and *"End of file"* should be handled in [PassportReader.swift:L361-L364](https://github.com/AndyQ/NFCPassportReader/blob/3a1a07812ceb2656513c8aa4a19076ade2d1c139/Sources/NFCPassportReader/PassportReader.swift#L361-L364).

nikolindima, 2020-12-20:
@smlu I try to read Russian passport. 
[SM] [0x0C, 0xB0, 0x00, 0x02, 0x0D, 0x97, 0x01, 0x00, 0x8E, 0x08, 0x3F, 0x65, 0x55, 0x2E, 0xD9, 0xB5, 0x34, 0xB5, 0x00]
Error reading tag: sw1 - 0x6B, sw2 - 0x00
reason: Wrong parameter(s) P1-P2]
ERROR - Wrong parameter(s) P1-P2]

[SM] [0x0C, 0xA4, 0x02, 0x0C, 0x15, 0x87, 0x09, 0x01, 0x82, 0xEE, 0xD4, 0x2F, 0x4B, 0xD7, 0x62, 0x31, 0x8E, 0x08, 0xCC, 0xB3, 0xB5, 0xF0, 0x63, 0xCD, 0x88, 0x85, 0x00]
Error reading tag: sw1 - 0x69, sw2 - 0x88
reason: SM data objects incorrect
ERROR - SM data objects incorrect

Error reading tag: sw1 - 0x69, sw2 - 0x85
reason: Conditions of use not satisfied

smlu, 2020-12-21:
You get interesting and unexpected error for the read binary command. From the first raw APDU, the `SW=0x6B00` error indicates that the offset `0x2` is somehow wrong. Do you maybe know what file are you trying to read? If you're able to read [COM](https://github.com/AndyQ/NFCPassportReader/blob/6500ce66ea17ddbdb192537a851e87be4a5a2829/Sources/NFCPassportReader/DataGroupParser.swift#L115) file and the failed file you're trying to read is any of the data group files (DG1-DG16), could you check if COM file contains it? This will determine if DG file exist on the passport.
For errors you get after executing the other 2 commands I'm guessing that the secure session is invalid. Is the BAC session not re-established after getting first error?

Since you already figured out that setting `maxDataLengthToRead` to `255` fixes your problem, could you check 2 more things.
First check if setting `maxDataLengthToRead` to `224` works for you. And second, could you add `|| errMsg.hasPrefix( "Wrong parameter" )` in this [if statement](https://github.com/AndyQ/NFCPassportReader/blob/6500ce66ea17ddbdb192537a851e87be4a5a2829/Sources/NFCPassportReader/PassportReader.swift#L361) so the max read var is changed to smaller length when you hit above `SW=0x6B00`.

nikolindima, 2020-12-22:
@smlu This error come before try to read special data group.

tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x281b721f0>)
Starting Basic Access Control (BAC)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): 18FA45BAF45D670BAF08CFBBEAFC3BDBD01401E8
Take the most significant 16 bytes to form the Kseed
	Kseed: 18FA45BAF45D670BAF08CFBBEAFC3BDB
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: 18FA45BAF45D670BAF08CFBBEAFC3BDB00000001
	Calculate the SHA-1 hash of D
		Hsha1(D): 0B88C46F919A238CE2F59FF900835104208DF4E7
	Form keys Ka and Kb
		Ka: 0B88C46F919A238C
		Kb: E2F59FF900835104
	Adjust parity bits
		Ka: 0B89C46E919B238C
		Kb: E3F49EF801835104
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: 18FA45BAF45D670BAF08CFBBEAFC3BDB00000002
	Calculate the SHA-1 hash of D
		Hsha1(D): 795E106A9308064A44B49866F9B5E725A50B15A8
	Form keys Ka and Kb
		Ka: 795E106A9308064A
		Kb: 44B49866F9B5E725
	Adjust parity bits
		Ka: 795E106B9208074A
		Kb: 45B59867F8B5E625
DATA - [212, 37, 194, 120, 117, 171, 83, 88]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: D425C27875AB5358
Generate an 8 byte random and a 16 byte random
	RND.IFD: E967A18B0121D325
	RND.Kifd: 46808FAD6EB51B80B3E070F3B8822A5A
Concatenate RND.IFD, RND.ICC and Kifd
	S: E967A18B0121D325D425C27875AB535846808FAD6EB51B80B3E070F3B8822A5A
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: F7276B8D102E8036895C7836836BF1CDD60F3F6FBF4FD6933EB8311073A4B24E
x0: F7276B8D102E8036
y0: 596E3CF5F066B411
x1: 895C7836836BF1CD
y1: 13EA39A8B0011BD0
x2: D60F3F6FBF4FD693
y2: E4AE39110AF9CD20
x3: 3EB8311073A4B24E
y3: 4B5DC3D3E5F06F80
x4: 8000000000000000
y4: F7711A3E487BE519
y: F7711A3E487BE519
bkey: 45B59867F8B5E625
akey: 795E106B9208074A
b: 66E49332028FB6A5
a: ED4F466F312F6C65
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: ED4F466F312F6C65
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: F7276B8D102E8036895C7836836BF1CDD60F3F6FBF4FD6933EB8311073A4B24EED4F466F312F6C65
DATA - [45, 36, 131, 155, 223, 186, 91, 40, 119, 165, 54, 189, 183, 158, 106, 52, 213, 176, 208, 15, 79, 25, 132, 123, 70, 151, 176, 29, 20, 120, 122, 44, 63, 14, 235, 203, 143, 81, 183, 182]
Decrypt and verify received data and compare received RND.IFD with generated RND.IFD 795E106B9208074A45B59867F8B5E625
Calculate XOR of Kifd and Kicc
	Kseed: D82331BFA72D7A586E1220340F0AC4F1
	Concatenate Kseed and c
		D: D82331BFA72D7A586E1220340F0AC4F100000001
	Calculate the SHA-1 hash of D
		Hsha1(D): 0D9898FE17158DA261851D3F4F84926D3B654556
	Form keys Ka and Kb
[... 248 lines cut ...]
	Eifd: 4BFE6CCE308364B9D9420F466C713B6B0C274663B04D7AEC8AF9E9CF848E8BBC
x0: 4BFE6CCE308364B9
y0: A8F0539EF4768C09
x1: D9420F466C713B6B
y1: 0BE6334B71B214C6
x2: 0C274663B04D7AEC
y2: EA799DF87DEE5BFA
x3: 8AF9E9CF848E8BBC
y3: D0EE662876C2516A
x4: 8000000000000000
y4: E97227854DB4E184
y: E97227854DB4E184
bkey: 45B59867F8B5E625
akey: 795E106B9208074A
b: FA4260DF63869F64
a: 2E438681BBD09A0E
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: 2E438681BBD09A0E
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: 4BFE6CCE308364B9D9420F466C713B6B0C274663B04D7AEC8AF9E9CF848E8BBC2E438681BBD09A0E
Error reading tag: sw1 - 0x69, sw2 - 0x85
reason: Conditions of use not satisfied
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 0.)
BAC Failed
tagReaderSession:didInvalidateWithError - Error Domain=NFCError Code=200 "Session invalidated by user" UserInfo={NSLocalizedDescription=Session invalidated by user}

nikolindima, 2020-12-22:
@smlu maxDataLengthToRead - works any value 11<value<256

I add || errMsg.hasPrefix( "Wrong parameter" )

This is result

tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x2826f9d40>)
Starting Basic Access Control (BAC)
Calculate the SHA-1 hash of MRZ_information
	Hsha1(MRZ_information): 18FA45BAF45D670BAF08CFBBEAFC3BDBD01401E8
Take the most significant 16 bytes to form the Kseed
	Kseed: 18FA45BAF45D670BAF08CFBBEAFC3BDB
Calculate the Basic Acces Keys (Kenc and Kmac) using Appendix 5.1
Compute Encryption key (c: 00000001
	Concatenate Kseed and c
		D: 18FA45BAF45D670BAF08CFBBEAFC3BDB00000001
	Calculate the SHA-1 hash of D
		Hsha1(D): 0B88C46F919A238CE2F59FF900835104208DF4E7
	Form keys Ka and Kb
		Ka: 0B88C46F919A238C
		Kb: E2F59FF900835104
	Adjust parity bits
		Ka: 0B89C46E919B238C
		Kb: E3F49EF801835104
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: 18FA45BAF45D670BAF08CFBBEAFC3BDB00000002
	Calculate the SHA-1 hash of D
		Hsha1(D): 795E106A9308064A44B49866F9B5E725A50B15A8
	Form keys Ka and Kb
		Ka: 795E106A9308064A
		Kb: 44B49866F9B5E725
	Adjust parity bits
		Ka: 795E106B9208074A
		Kb: 45B59867F8B5E625
DATA - [74, 107, 23, 54, 224, 163, 29, 127]
Request an 8 byte random number from the MRTD's chip
	RND.ICC: 4A6B1736E0A31D7F
Generate an 8 byte random and a 16 byte random
	RND.IFD: C1ACA522F5B5558D
	RND.Kifd: 06402E2321AA56E49594B75A753D5A58
Concatenate RND.IFD, RND.ICC and Kifd
	S: C1ACA522F5B5558D4A6B1736E0A31D7F06402E2321AA56E49594B75A753D5A58
Encrypt S with TDES key Kenc as calculated in Appendix 5.2
	Eifd: 25AF3D159EA7BF1CE279D07E83C043840590917AE5CF026BB62B3605E4F4AA3F
x0: 25AF3D159EA7BF1C
y0: 662CB3FDAFA8F1C1
x1: E279D07E83C04384
y1: 0FDA028E9DAE76D8
x2: 0590917AE5CF026B
y2: FF207CE133655530
x3: B62B3605E4F4AA3F
y3: D45AB06B13420FCD
x4: 8000000000000000
y4: 253409B61E3C3EF0
y: 253409B61E3C3EF0
bkey: 45B59867F8B5E625
akey: 795E106B9208074A
b: 59A4357CD37E6E06
a: 4C26323BFD42F3CB
Compute MAC over eifd with TDES key Kmac as calculated in-Appendix 5.2
	Mifd: 4C26323BFD42F3CB
Construct command data for MUTUAL AUTHENTICATE
	cmd_data: 25AF3D159EA7BF1CE279D07E83C043840590917AE5CF026BB62B3605E4F4AA3F4C26323BFD42F3CB
DATA - [58, 212, 211, 198, 249, 123, 245, 21, 104, 217, 207, 233, 212, 35, 2, 36, 89, 94, 193, 214, 192, 118, 23, 241, 65, 16, 154, 46, 118, 149, 17, 99, 131, 6, 122, 18, 6, 182, 220, 199]
Decrypt and verify received data and compare received RND.IFD with generated RND.IFD 795E106B9208074A45B59867F8B5E625
Calculate XOR of Kifd and Kicc
	Kseed: F0DC8E5C2B86622DD5C0F70D0D76BE32
	Concatenate Kseed and c
[... 182 lines cut ...]
	Concatenate Kseed and c
		D: 18FA45BAF45D670BAF08CFBBEAFC3BDB00000001
	Calculate the SHA-1 hash of D
		Hsha1(D): 0B88C46F919A238CE2F59FF900835104208DF4E7
	Form keys Ka and Kb
		Ka: 0B88C46F919A238C
		Kb: E2F59FF900835104
	Adjust parity bits
		Ka: 0B89C46E919B238C
		Kb: E3F49EF801835104
Compute MAC Computation key (c: 00000002
	Concatenate Kseed and c
		D: 18FA45BAF45D670BAF08CFBBEAFC3BDB00000002
	Calculate the SHA-1 hash of D
		Hsha1(D): 795E106A9308064A44B49866F9B5E725A50B15A8
	Form keys Ka and Kb
		Ka: 795E106A9308064A
		Kb: 44B49866F9B5E725
	Adjust parity bits
		Ka: 795E106B9208074A
		Kb: 45B59867F8B5E625
Error reading tag: sw1 - 0x69, sw2 - 0x88
reason: SM data objects incorrect
ERROR - The operation couldn’t be completed. (NFCPassportReader.TagError error 0.)
BAC Failed

smlu, 2020-12-28:
@nikolindima thanks for all provided logs and infos. I went through all the logs and I can't spot any part in the log that would suggest error in the implementation. I also couldn't find any reference on the internet or in the ISO7816 and ICAO9303 eMRTD specs that would suggests for SW=`6B00` any other error than wrong P1-P2 parameters. 
I found only 1 reference in [ICAO Radio Frequency and Protocol Testing part 4](https://www.icao.int/Security/FAL/TRIP/Documents/TR%20-%20RF%20and%20Protocol%20Testing%20Part%204%20V2.11.pdf), page 63, section *7.6.4 ISO7816_F_04: Reading end of file with status word 6B00*, that suggest ICC could return SW=`6B00` when reading beyond the EOF. But the test case doesn't specify if the returned SW=`6B00` due to wrong offset (P1-P2 params) or max read (Le param).

Based on this, I'm guessing that your passport doesn't support arbitrary data read for READ BINARY command, as it has been already found out.

The other interesting thing I've noticed from the logs is that your passport handles session termination differently than specified in ISO7816 when an error occurs. The ISO716 specs say that session should be terminated on both sides as soon as any kind of error occurs, but from the provided logs it looks like your passport stays in limbo until you hit `SM` error (on second attempt to read EF.COM) and then terminates active session. I haven't figured out why you get SW=`6985` on the second attempt to establish BAC session after session termination. It could be passport's fault (wrong ISO7816 implementation) or there is something I'm missing in this implementation that should be done differently. In any case it's strange that the first BAC session establishment goes through without a problem but fails on the second attempt.

smlu, 2020-12-28:
@AndyQ I'd suggest the variable [maxDataLengthToRead](https://github.com/AndyQ/NFCPassportReader/blob/6500ce66ea17ddbdb192537a851e87be4a5a2829/Sources/NFCPassportReader/TagReader.swift#L190) is changed to `224` bytes (same value is used in JMRTD library) as arbitrary data read support can't be for sure determined for every passport. I encountered at least 1 more passport (Canada) that couldn't handle the arbitrary data read (Got SW=`6700` there).

And I'd also suggest to maybe implement a logic to re-try one more time to establish BAC session if SW=`6988` occurs. The above logs suggest that there could be passports that don't correctly terminate session on any kind of error but only on SW=`6988`.

AndyQ, 2020-12-28:
@smlu - Thanks will take a look.  Made a couple of changes in the last release around data lengths as was getting a number of reports of some older and newer passports not working correctly.  I tried a number of various lengths - not sure if 224 was one of them though - and had a lot of issues with reads failing.  Maybe start with 256, then drop down to 224 if a 6700 error, then finally 160?

I'll also check what I do on a 6988 error - I thought I tried to restablish BAC on any error up to 3 times but.....

nikolindima, 2020-12-28:
@smlu @AndyQ other libraries use 112 size

smlu, 2020-12-28:
@AndyQ that would probably be a proper way of handling 6700 error for the most of passports (start at 256 and then drop down to lower number on error). But in @nikolindima case it won't help because the return error is 6B00 instead of 6700 and the phone can't  re-establish new session with passport after it hits an error (at least above logs indicates so and I don't see any error in the BAC implementation). I can't think of any good solution for this case other than storing smaller `maxDataLengthToRead` value when error occurs and instruct user to re-connect phone to the passport. Because of this I believe the better solution would be to use smaller `maxDataLengthToRead` value to begin with and not trying with arbitrary data read at all. This should probably cover most of the cases.

@nikolindima in what libraries have you seen 112 bytes for read data block size?

nikolindima, 2020-12-28:
@smlu this one https://github.com/rgex/bondi-passport-reader-ios

AndyQ, 2020-12-29:
So @nikolindima , could you please try changing TagReader.swift (around line 190) to:
var maxDataLengthToRead : Int = 160

Let me know if that works ok for you?

nikolindima, 2020-12-29:
@AndyQ i try maxDataLengthToRead - works any value 11<value<256

AndyQ, 2021-02-01:
Thanks - the next version will be reverting back to 160 for the defaul (although you will be able to override this if you really want to).

closed 2021-02-01 by AndyQ