AndyQ/NFCPassportReader #90: OpenSSL 1.1.x support?

lendvair, 2021-01-07:
Hi @AndyQ,

Are you planning to use the new OpenSSL-Universal dependency which is updated to OpenSSL 1.1.1i? There are many API breaking changes in this, but the older one (1.0.2) is out of support. 

Best regards,
Richard

AndyQ, 2021-01-18:
Yes indeed - I'm hoping to start to look at that soon.

AndyQ, 2021-02-15:
So, I've made a fair bit of progress on this - all the code changes are done and just testing - I'll check in a new branch soon for testing.

I've also made it a Swift Package so it should work with SPM.

It will probably go live with the next release - 1.1.0 - its already got breaking changes in it so this is probably a good time to also upgrade to OpenSSL 1.1

AndyQ, 2021-02-22:
This is now avavailable in branch 1.1.0 - please test and let me know if any issues!

closed 2021-02-22 by AndyQ