AndyQ/NFCPassportReader #91: maxDataLengthToRead in TagReader

mlorenzo-alice, 2021-01-20:
Hello, 

First of all, thank you for your work.

I am trying to read de Spanish ID but got the following error. I have managed to read the Spanish passport successfully. The ReadID app was able to read both.

```
Construct and send protected APDU
	ProtectedAPDU: 0CB000020D9701008E08E554A4B519D5B64D00
[SM] [0x0C, 0xB0, 0x00, 0x02, 0x0D, 0x97, 0x01, 0x00, 0x8E, 0x08, 0xE5, 0x54, 0xA4, 0xB5, 0x19, 0xD5, 0xB6, 0x4D, 0x00]
Error reading tag: sw1 - 0x62, sw2 - 0x82
reason: End of file/record reached before reading Le bytes
ERROR - End of file/record reached before reading Le bytes
Starting Basic Access Control (BAC)
```

Searching in some other issues, I finally replaced the `maxDataLengthToRead` from 256 to 160. This solved my issue.

How can I set this lenght to work for both IDs and passports?

Thank you in advance.

Regards.

AndyQ, 2021-02-01:
160 should work fine for both passports and IDs.  In fact the next version will be reverting back to 160 (even though some passports support 256 (arbitrary lengths - basically just read until the passport returns no more data), a large number don't and trying to drop back down causes other issues.  So the default will be 160 (but you will be able to override that if you really want to.

AndyQ, 2021-02-05:
This has now been released so I'm closing - please re-open or raise a new request if you are still having issues!

closed 2021-02-05 by AndyQ