AndyQ/NFCPassportReader #94: iOS 14 connection issues

haeser, 2021-02-10:
Hi,

some time ago there was a discussion related to iOS 14 (https://github.com/AndyQ/NFCPassportReader/issues/74).

We are currently observing something similar. For a couple of devices, reading the passport worked perfectly fine in the past. It no more works on 14.2, 14.3, 14.4.
Same configuration (e.g. 14.4 and same model) on a different phone reads the passport just fine. We can rule out that the chip is broken.
Sometimes, after restarting the phone the first scan works, all subsequent scan fail. Another observation: before restarting the phone the app needs to be force closed.

The connection issues occur very random sometimes when connecting to the tag, some other times during DG reads. All leading to the "stack error".

Did anyone else observe similar behavior and maybe has a solution for it?

[Attached a complete log](https://github.com/AndyQ/NFCPassportReader/files/5959798/console%2Bnfcpassportreaderapp.txt). I think the relevant part is in this snippet. The log and the snippet are from an Xr running 14.2

```
default	11:15:57.679886-0500	SharingViewService	Scan ViewDidAppear
default	11:15:58.047046-0500	nfcd	00000001 0b027840 -[_NFReaderSession handleRemoteTagsDetected:]:532  1 tags found
default	11:15:58.052084-0500	nfcd	00000001 0b027840 -[_NFReaderSession _sync_connect:callback:]:616  NFCPassportReaderApp (398) _NFReaderSession-634666557.001496
error	11:15:58.054877-0500	nfcd	Generic error received: PH_NCINFC_STATUS_DISCOVERY_TARGET_ACTIVATION_FAILED
error	11:15:58.055166-0500	nfcd	Generic error received: phNciNfc_e_DiscTgtActvnFailed
error	11:15:58.055253-0500	nfcd	NtfMFC: Generic error received: PH_NCINFC_STATUS_DISCOVERY_TARGET_ACTIVATION_FAILED
error	11:15:58.055325-0500	nfcd	phLibNfc_RemoteDev_Connect_Cb: Lower layer has returned failure status
error	11:15:58.055370-0500	nfcd	NFDriverRemoteDevConnect:6379  phLibNfc_RemoteDev_Connect failed: 0x00ff
error	11:15:58.055416-0500	nfcd	00000001 09f09000 -[NFDriverWrapper connectTag:]:1475  Failed to connect to tag: <NFTagInternal: 0x10b027190>-5BBC9D00A40B5EF2 { Tech=B Type=Generic 4B ID=04D7B5DFE1F35E117781A5, NDEF=0, capacity=0, messageSize=0 }
error	11:15:58.056196-0500	NFCPassportReaderApp	00000002 81cad0e0 -[NFCTagReaderSession _connectTag:error:]:621  Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x28209dc50 {Error Domain=nfcd Code=15 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error}}}
default	11:15:58.057828-0500	nfcd	00000001 0b027840 -[_NFReaderSession endSession:]:205  _NFReaderSession-634666557.001496
default	11:15:58.058757-0500	nfcd	00000001 0b027840 -[_NFReaderSession internalEndSession:]:188  _NFReaderSession-634666557.001496
default	11:15:58.059473-0500	nfcd	00000001 09f08590 -[_NFHardwareManager _dequeueHelperReaderSession:]:1960  Suspend until UI invalidation handler executes
default	11:15:58.059705-0500	nfcd	00000001 0b027840 -[_NFReaderSession didEndSession:]:155  NFCPassportReaderApp (398) ended _NFReaderSession-634666557.001496
```

AndyQ, 2021-02-12:
Which version are you using?

haeser, 2021-02-12:
running from master. We did some more testing with different devices and think it is an issue related to the Xr model running 14.x with US passports only. Other passports work just fine.

daniel-mohemian, 2021-02-23:
After some more testing we observed that iOS 14.5 beta 2 seems to fix this issue.

haeser, 2021-05-12:
the issue is no more observed in 14.5, which is believed to have a wide adoption already. Closing the issue.

closed 2021-05-12 by haeser

arunlj561, 2021-06-05:
This issue is again open in 14.6, we get connection issue on 14.6 but not on 14.5.