AndyQ/NFCPassportReader #95: Crash on ChipAuthentication

filom, 2021-03-03:
In the wrapDO the `arr` array is 256 which of course cannot fit in an UInt8.
```
func wrapDO( b : UInt8, arr : [UInt8] ) -> [UInt8] {
    let result : [UInt8] = [b, UInt8(arr.count)] + arr
    return result;
}
```

I believe that `wrapDO` and `unwrapDO` are ASN.1 encoded data, right ? If so length of content has an different encoding to allow size > 255. 

```
func wrapDO( b : UInt8, arr : [UInt8] ) -> [UInt8] {
    let result : [UInt8] = [b] + lengthHeader(of: arr.count) + arr
    return result
}

func lengthHeader(of size: Int) -> [UInt8] {
    if size > 127 {
        var lenBytes = byteArray(from: size)
        while lenBytes.first == 0 { lenBytes.removeFirst() }
        let len: UInt8 = 0x80 | UInt8(lenBytes.count)
        return [len] + lenBytes
    } else {
        return [UInt8(size)]
    }
}

func byteArray<T>(from value: T) -> [UInt8] where T: FixedWidthInteger {
    return withUnsafeBytes(of: value.bigEndian, Array.init)
}
```

The `unwrapDO` don't really need to know the size, but just the size of the length header:
```
func unwrapDO( tag : UInt8, wrappedData : [UInt8]) throws -> [UInt8] {
    let actualTag = wrappedData[0];
    if actualTag != tag {
        throw NFCPassportReaderError.InvalidASN1Value
    }
    var headerSize = 1
    if (wrappedData[1] & 0x80) != 0 {
        headerSize += 1 + Int(wrappedData[1] - 0x80)
    } else {
        headerSize += 1
    }
    let result = [UInt8](wrappedData[headerSize...])
    return result
}
```

This seems to fix the wrap problem but then `NFCISO7816APDU(data:Data(protectedAPDU))!` return nil and crash.

AndyQ, 2021-03-06:
Thats great thanks! - I'll look at getting that in the next version.

Btw - it would be really useful to know at which point did the buffer go over 256 bytes?  I'm assuming this is when sending the public key?
I've been testing and the largest amount of data I've seen being sent down is about 93 bytes when the public key down, even if we are sending multiple chunks of data they are chunked into groups of 224 bytes (which with the additional header shouldn't take it over 256).

filom, 2021-03-06:
Yes, it happens on `sendPublicKey` in case of "DESede":
`let wrappedKeyData = wrapDO( b:0x91, arr:keyData)` <- keyData is 256 length.

Here the key structure [DG14]:
```
pos:0, d=0, hl=4, l=882: cons: SET 
pos:4, d=1, hl=4, l=826: cons: SEQUENCE 
pos:8, d=2, hl=2, l=9: prim: OBJECT 0.4.0.127.0.7.2.2.1.1
pos:19, d=2, hl=4, l=811: cons: SEQUENCE 
pos:23, d=3, hl=4, l=541: cons: SEQUENCE 
pos:27, d=4, hl=2, l=9: prim: OBJECT dhKeyAgreement
pos:38, d=4, hl=4, l=526: cons: SEQUENCE 
pos:42, d=5, hl=4, l=257: prim: INTEGER 
pos:303, d=5, hl=4, l=257: prim: INTEGER 
pos:564, d=5, hl=2, l=2: prim: INTEGER E0
pos:568, d=3, hl=4, l=262: prim: BIT STRING 
pos:834, d=1, hl=2, l=15: cons: SEQUENCE 
pos:836, d=2, hl=2, l=10: prim: OBJECT 0.4.0.127.0.7.2.2.3.1.1
pos:848, d=2, hl=2, l=1: prim: INTEGER 01
pos:851, d=1, hl=2, l=13: cons: SEQUENCE 
pos:853, d=2, hl=2, l=8: prim: OBJECT 0.4.0.127.0.7.2.2.2
pos:863, d=2, hl=2, l=1: prim: INTEGER 01
pos:866, d=1, hl=2, l=18: cons: SEQUENCE 
pos:868, d=2, hl=2, l=10: prim: OBJECT 0.4.0.127.0.7.2.2.4.1.1
pos:880, d=2, hl=2, l=1: prim: INTEGER 02
pos:883, d=2, hl=2, l=1: prim: INTEGER 02
```

AndyQ, 2021-03-06:
Really interesting - I didn't see anything that big on my passports!
I'd actually already changed the wrapDO/unwrapDO functions to use Apple's   TKBERTLVRecord encoder which also handles the encoding better than me doing it manually ;) and I've also added your fix in.

This will be in the next release (along with an initial PACE implemention).

Thanks again!

AndyQ, 2021-03-07:
Also @filom i noticed that your passports uses DH keys in PACE - would you be willing to help test that?  (I don't have a passport that supports that)

filom, 2021-03-07:
Yes, no problem. How can I test it ?

AndyQ, 2021-03-07:
Any chance you could drop me an email (address in profile)?

AndyQ, 2021-03-12:
This should be fixed in 1.1.2 which is now released

closed 2021-03-12 by AndyQ

tienngx, 2021-07-09:
Hello AndyQ
We can't do with PACE CAN. Do you have any suggestions for this type?
Thanks & Best Regards

AndyQ, 2021-07-09:
Hi - there is a pull request available which adds support for PACE CAN.

I haven't yet merged it as I have currently don't have access to any documents that support CAN so I'm unable to test it but it is on my list of things to do!