AndyQ/NFCPassportReader #96: calcHash returns NFCPassportReaderError.InvalidHashAlgorithmSpecified, because of sha224

wellbranding, 2021-03-10:
Hello, we are testing your library functionality with Lithuanian Passports. It does extract all information, but displays the following error message: Couldn't verify SOD signature. 
I added breakpoints and looked at the source code and discovered the exact error place.
The error occurs in the calcHash method because hashAlgorithm contains a **sha224** value. 

Why this value is not supported and what can we do about it? 

I can provide all information related to the exact document if it might help you.

This is really important for us :)

AndyQ, 2021-03-11:
Hi, if you are happy to export the passport data and send it to me I'll be more than happy to add support - my email address is in my profile.  Please only export the non-personal identifying data at this time though please.

AndyQ, 2021-03-11:
I've added support for SHA224 hashes and these will be in the next release.

AndyQ, 2021-03-12:
This should be fixed in 1.1.2 which is now released

closed 2021-03-12 by AndyQ