AndyQ/NFCPassportReader #97: EVP_DigestVerifyFinal failed, sha224

wellbranding, 2021-03-15:
Hello, thanks for providing support for the sha224 [in the following issue](https://github.com/AndyQ/NFCPassportReader/issues/96). I tested one more time, and indeed calcHash no longer fails. However, the SOD verification still fails. Right now the issue is in the **verifySignature** method. It fails at the last part, where **EVP_DigestVerifyFinal** is called. The returned nRes is -1 as you can see in the screenshot.
[image]

Also, I tried to call the verifyPassport method with the useCMSVerification set **as true**. It still did not work, and the error was thrown in the following piece of code:
[image]

Again, if you need any Passport data, I would definitely provide it.
Sincerely, Viktor

AndyQ, 2021-03-15:
Hi - if you could email me the passport data (address in profile) - export through the sample app) then I'll check it out and see whats going on!

Thanks!

AndyQ, 2021-03-16:
Thanks for helping identify and test the fix - will be out in the next release

AndyQ, 2021-03-24:
This should now be fixed in the 1.1.3 release

closed 2021-03-24 by AndyQ