AndyQ/NFCPassportReader #98: Will it work for ID cards. Does ID cards also follow same protocols like BAC and PACE?

RaknasAvis, 2021-03-22:
I would like to read data from the ID cards which are having NFC chips. Is it completely different from passports?
Any links would be appreciated.

AndyQ, 2021-03-22:
It should - its untested but I think they use the same AID. PACE may not yet be supported (I have currently only implemented MRZ keys and I *think* ID cards may use CAN but not sure), but BAC should work.

typelogic, 2021-03-26:
> I would like to read data from the ID cards which are having NFC chips. Is it completely different from passports?
> Any links would be appreciated.

There are 12 of these, but you may start [here](https://www.icao.int/publications/documents/9303_p10_cons_en.pdf)

closed 2023-01-25 by AndyQ