AndyQ/NFCPassportReader #100: Incomplete inferOID

typelogic, 2021-03-26:
The current implementation of [inferOID](https://github.com/AndyQ/NFCPassportReader/blob/master/Sources/NFCPassportReader/ChipAuthenticationHandler.swift#L110-L121) looks incomplete. The [9303_p11_cons_en.pdf](https://imgur.com/BCckoMX.png) seems to describe the complete listing. 

How do we approach to correctly infer the OID when `ChipAuthenticationInfo` is missing? The JMRTD's [inferChipAuthenticationOIDfromPublicKeyOID](https://sourceforge.net/p/jmrtd/code/HEAD/tree/trunk/jmrtd/src/main/java/org/jmrtd/protocol/EACCAProtocol.java) method is also incomplete.

AndyQ, 2021-03-26:
Not sure what you mean by incomplete.  The ChipAuthenticationInfo contains the details of what implementation should be used.
Some passports for some reason don't include this so there is no way of knowing which is actually used, so we need to take a guess.
There are two different types:  DH and ECDSA, and within them you can use both 3DES and AES based encryption.  If AES is used, then there are different key lengths.

However, however the Object ID for the Public Key only tells us that we are using either DH or ECDSA so we guess for the rest and based on the passports I've seen, in these cases they are all 3DES based so these are the ones I pick (as does JMRTD).

However, if there is a more complete way of guessing then I'd really like to know!

typelogic, 2021-03-30:
I have one remote (not mine) epassport that failed on `Chip Authentication` using JMRTD's infer method. What do you think of looping through each until it succeeds?

AndyQ, 2021-03-30:
Would be interested to know what the type it actually was using was!  If you are able to share the DG14 element and also the nationality of that passport that may be useful (not sure though). 

The only issue with try each one if we are inferring, then it would probably break the secure session - passports tend to abort the session on an error with no warning which means that you would need to re-establish BAC each time adding a fair bit of complexity.

typelogic, 2021-03-31:
I'm able to grab DG14 [here](https://gist.github.com/typelogic/351d2c5fc62f65dcaf063618e1855f5a).

typelogic, 2021-04-01:
> Would be interested to know what the type it actually was using was! If you are able to share the DG14 element and also the nationality of that passport that may be useful (not sure though).
>
> The [...]

According to an [ICAO protocol specification](https://imgur.com/aHDkAb3.png), it won't harm the state of the secure messaging if ever on failed to do `Chip Authentication`.

typelogic, 2021-04-01:
From [this ICAO spec](https://imgur.com/2Mtb4Xr.png), isn't epassports with missing `ChipAuthenticationInfo` be considered deviants? However, he was able to do `CA` using `ReadID` so I wonder how `ReadID` did it.

typelogic, 2021-04-05:
I just tested using JMRTD that looping through each possible oid does not reset the secure messaging, until it finds the correct oid and is able to do a successful `doEACCA()`. In light of this result and [what is stated here](https://imgur.com/2Mtb4Xr.png), I think I'll stick on my looping technique rather than choosing one. For example, [other possible possible of OID](https://github.com/jllarraz/AndroidPassportReader/issues/15) is cited here.

AndyQ, 2021-04-05:
If you want to submit a pull request I'm happy to see if I can get this tested on some of the passports I've found that also have the missing ChipAutheniticationInfo with a view to implementing it in a future version if all works ok.

danydev, 2023-02-04:
@typelogic I'm curious which country passports did have this problem?