AndyQ/NFCPassportReader #101: Session invalidated

xtravanta, 2021-04-09:
Hi,

I tried your example code and on both the documents i tried it stops on reading dg2 with the message Session invaidated.
Do you have any idea what this could be? 

The documents are scanned are Dutch passports.

see log below:

```
QKMRZScanner.QKMRZScanResult
tagReaderSessionDidBecomeActive
tagReaderSession:didDetect - iso7816(<NFCISO7816Tag: 0x28212e880>)
tagReaderSession:connecting to tag - iso7816(<NFCISO7816Tag: 0x28212e880>)
tagReaderSession:connected to tag - starting authentication
TagReader - Number of data bytes to read - 20
Read CardAccess - data 31143012060A04007F0007020204020402010202010E
Starting Password Authenticated Connection Establishment (PACE)
Performing PACE with id-PACE-ECDH-GM-AES-CBC-CMAC-256
Doing PACE Step1...
Doing PACE Step2...
   Using General Mapping (GM)...
Generating ECDH mapping keys from parameterSpec - 929
Sending public mapping key to passport..
Received passports public mapping key
Doing ECDH Mapping agreement
Doing PACE Step3 - Key Exchange
Generated Ephemeral key pair
Sending ephemeral public key to passport
Doing PACE Step3 Key Agreement...
Computing shared secret...
Deriving ksEnc and ksMac keys from shared secret
Generating authentication token
Sending auth token to passport
Auth token from passport matches expected token!
Restarting secure messaging using AES encryption
PACE Successful
Re-selecting eMRTD Application
Reading tag - COM
TagReader - Number of data bytes to read - 23
DG Found - ["DG1", "DG2", "DG15", "DG3", "DG14"]
Reading tag - DG14
TagReader - Number of data bytes to read - 473
Performing Chip Authentication - number of public keys found - 1
Starting Chip Authentication!
Public Key successfully sent to passport!
Restarting secure messaging using AES encryption
Finished Chip Authentication - success - true
Reading tag - SOD
TagReader - Number of data bytes to read - 2660
Reading tag - DG1
TagReader - Number of data bytes to read - 91
Reading tag - DG2
TagReader - Number of data bytes to read - 15075
TagReader - Error reading tag - Session invalidated)
ERROR - Session invalidated
Starting Basic Access Control (BAC)
BACHandler - deriving Document Basic Access Keys
BACHandler - Getting initial challenge
TagReader - Error reading tag - Session invalidated)
ERROR - Session invalidated
BAC Failed
tagReaderSession:didInvalidateWithError - Session invalidated by user
```

igashev, 2021-05-10:
Hi, did you manage to work around this? Or is there a solution to it?

AndyQ, 2023-02-03:
See #107

closed 2023-02-03 by AndyQ