AndyQ/NFCPassportReader #103: I got issue with XCode 12.3

coffteam2020, 2021-04-20:
Undefined symbol: enum case for (extension in Foundation):__C.NSScanner.NumberRepresentation.decimal((extension in Foundation):__C.NSScanner.NumberRepresentation.Type) -> (extension in Foundation):__C.NSScanner.NumberRepresentation

Undefined symbol: type metadata accessor for (extension in Foundation):__C.NSScanner.NumberRepresentation

Undefined symbol: (extension in Foundation):__C.NSScanner.scanInt(representation: (extension in Foundation):__C.NSScanner.NumberRepresentation) -> Swift.Int?

Undefined symbol: (extension in Foundation):__C.NSScanner.scanUpToCharacters(from: Foundation.CharacterSet) -> Swift.String?

Undefined symbol: nominal type descriptor for (extension in Foundation):__C.NSNotificationCenter.Publisher

Undefined symbol: protocol conformance descriptor for (extension in Foundation):__C.NSNotificationCenter.Publisher : Combine.Publisher in Foundation

Undefined symbol: type metadata accessor for (extension in Foundation):__C.NSNotificationCenter.Publisher

Undefined symbol: (extension in Foundation):__C.NSNotificationCenter.publisher(for: __C.NSNotificationName, object: Swift.AnyObject?) -> (extension in Foundation):__C.NSNotificationCenter.Publisher

Thank for this project. Could you help me out? @AndyQ @WooD1k @ruudk @danydev @ChristianNorbertBraun

sbycrosz, 2021-04-26:
Hi @coffteam2020, did you manage to figure this out?

AndyQ, 2021-04-27:
Do you have any more information as to where you see this?  I've not seen this.

sbycrosz, 2021-05-06:
Hi, 

In our case, we found out that this is caused by an incompatible swift version

Removing  `"\"$(TOOLCHAIN_DIR)/usr/lib/swift-5.0/$(PLATFORM_NAME)\"",` from `LIBRARY_SEARCH_PATHS` would fix the compilation issue

- Sam

closed 2021-09-16 by AndyQ